SatDump/SatDump #834: GOES HRIT Data Decoder stuck idle

BudaLope, 2024-10-09:
With both Viterbi and the Deframer synced with an average SNR of ~6 the GOES HRIT Data Decoder produces no images. This exact setup has worked previously with an average SNR of about 4.5-5.

I found a [Reddit Thread](https://www.reddit.com/r/amateursatellites/comments/1fffgfc/cant_get_a_goes_image/) where someone had the same issue, but it's unresolved 

RTL-SDR Blog V4, Windows 10

v1.2.1-4bc20b2

[satdump-20241009T163728.log](https://github.com/user-attachments/files/17318522/satdump-20241009T163728.log)

There are no files in the live_output folder for debugging.

Things tried

~30 minutes of waiting for each
- Every sample rate on the RTL-SDR from 1.536 Msps and up
- AGC
- No AGC
- DC Blocking both on and off
- Save DCS LRIT and Save ALL LRIT, no output from either
- Removed all trace of Satdump from PC, and restarting
- Satdump version 1.1.4 (Version previously working)
- Satdump version 1.2.0
- Latest Nightly Satdump

[image]

Is 'Progress -nan(ind)%' a problem? I wish I had the logs from my good run to verify.

JVital2013, 2024-10-10:
Hello! Thanks for reaching out. We've had a few reports of this, and every time it's been because the user is pointed at GOES-19 instead of GOES-16 or GOES-18. GOES-19 is currently sending filler.

I suspect the same thing is happening here. Please verify your aim and try again.

Also,  'Progress -nan(ind)%' is not a problem. We plan to fix this (eventually)

BudaLope, 2024-10-10:
I'm very sure I am pointed at GOES-16, as setting up on a fresh windows install on an old laptop I had I am once again receiving images.
Dish hasn't moved, the only piece of the puzzle that's changed is the PC being used. Which points to it being an issue with my PC...
If you want to pursue this further I can rerun the cable and attempt to debug more.
[image]

JVital2013, 2024-10-14:
If you would like to pursue the issue and re-run the cable, it might be worthwhile. What would be helpful to try is:

1. Re-run it and ensure you are still having the problem on your other PC with the latest nightly build
2. If you are, does the constellation kind of look like it is "rotating"? This could be indicative of interference between GOES-16 and GOES-19
3. If you are still having issues, regardless of step 2, stop decoding and exit SatDump. Edit `C:\Program Files\SatDump\share\satdump\pipelines\GOES.json` and remove these two lines, from under goes_hrit/live_cfg/default

    [image]

4. Try receiving GOES again. This time, it will not decode images live, but instead save a cadu file. Let this run for 5-10 minutes.
5. Zip the cadu file and upload it here for analysis

BudaLope, 2024-10-17:
Sorry for the delay. I've rerun the setup and can no longer reproduce the effect. I'm still sure I wasn't pointing at GOES-19 as I haven't moved the dish the entire time.

But! Cannot reproduce, and my new setup is far more ergonomic so feel free to close this. Sorry for wasting your time!

closed 2024-10-17 by JVital2013

JVital2013, 2024-10-20:
No worries - I don't feel it was a time waste. If it happens again on any setup, feel free to follow the steps above so we can troubleshoot it.