SatDump/SatDump #836: Satdump network source

f4tnk, 2024-10-17:
Hi,

I'm trying to recover a UDP signal with satdump **(Branch git master)**

I use the following command :

satdump live noaa_apt $OUT --source net_source --source_id 0 --port $UDP_DUMP_PORT --satellite_number 15 --samplerate $SAMP --frequency $FREQ"

But the flow is not established and the image is empty.

Am I missing a setting? I saw in the code the possibility of configuring the udp mode but I don't know if I should select it.

If I enter an rtl sdr + device number , an image is generated.

Thank you in advance.

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
17/10/2024] (I)    _____       __  ____                      
[19:31:19 - 17/10/2024] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[19:31:19 - 17/10/2024] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[19:31:19 - 17/10/2024] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[19:31:19 - 17/10/2024] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[19:31:19 - 17/10/2024] (I)                                     /_/      
[19:31:19 - 17/10/2024] (I) Starting SatDump v1.2.1-47e5757
[19:31:19 - 17/10/2024] (I) 
[19:31:19 - 17/10/2024] (I) Loading config /usr/share/satdump/satdump_cfg.json
[19:31:19 - 17/10/2024] (I) Loading user settings /var/lib/satnogs-client/.config/satdump/settings.json
[19:31:20 - 17/10/2024] (I) Found devices (sources) :
[19:31:20 - 17/10/2024] (I) - File Source
[19:31:20 - 17/10/2024] (D)     - type : file
[19:31:20 - 17/10/2024] (D)     - id : 0
[19:31:20 - 17/10/2024] (I) - Network Source
[19:31:20 - 17/10/2024] (D)     - type : net_source
[19:31:20 - 17/10/2024] (D)     - id : 0
[19:31:20 - 17/10/2024] (I) - Realtek RTL2838UHIDIR #00002000
[19:31:20 - 17/10/2024] (D)     - type : rtlsdr
[19:31:20 - 17/10/2024] (D)     - id : 00002000
[19:31:20 - 17/10/2024] (I) - RTLSDRBlog Blog V4 #00003000
[19:31:20 - 17/10/2024] (D)     - type : rtlsdr
[19:31:20 - 17/10/2024] (D)     - id : 00003000
[19:31:20 - 17/10/2024] (I) - Realtek RTL2838UHIDIR #00001000
[19:31:20 - 17/10/2024] (D)     - type : rtlsdr
[19:31:20 - 17/10/2024] (D)     - id : 00001000
[19:31:20 - 17/10/2024] (I) - RTL-TCP
[19:31:20 - 17/10/2024] (D)     - type : rtltcp
[19:31:20 - 17/10/2024] (D)     - id : 0
[19:31:20 - 17/10/2024] (I) - SDR++ Server
[19:31:20 - 17/10/2024] (D)     - type : sdrpp_server
[19:31:20 - 17/10/2024] (D)     - id : 0
[19:31:20 - 17/10/2024] (I) - SpyServer
[19:31:20 - 17/10/2024] (D)     - type : spyserver
[19:31:20 - 17/10/2024] (D)     - id : 0
[19:31:20 - 17/10/2024] (I) Found devices (sinks) :
[19:31:20 - 17/10/2024] (I) Done! Goodbye

-------------------------------------------------------------------------------

[19:13:18 - 17/10/2024] (I)    _____       __  ____                      
[19:13:18 - 17/10/2024] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[19:13:18 - 17/10/2024] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[19:13:18 - 17/10/2024] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[19:13:18 - 17/10/2024] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[19:13:18 - 17/10/2024] (I)                                     /_/      
[19:13:18 - 17/10/2024] (I) Starting SatDump v1.2.1-47e5757
[19:13:18 - 17/10/2024] (I) 
[19:13:18 - 17/10/2024] (I) Loading config /usr/share/satdump/satdump_cfg.json
[19:13:18 - 17/10/2024] (I) Loading user settings /var/lib/satnogs-client/.config/satdump/settings.json
[19:13:18 - 17/10/2024] (I) Loading plugins from /usr/lib/x86_64-linux-gnu/satdump/plugins
[19:13:18 - 17/10/2024] (T) Loading plugin /usr/lib/x86_64-linux-gnu/satdump/plugins/libjpss_support.so...
[19:13:18 - 17/10/2024] (T) Plugin jpss_support loaded!
[... 642 lines cut ...]
[19:27:37 - 17/10/2024] (W) Signal Received. Stopping.
[19:27:37 - 17/10/2024] (I) Stop processing
[19:27:37 - 17/10/2024] (I) Demodulation finished
[19:27:37 - 17/10/2024] (I) Got 0 lines...
[19:27:37 - 17/10/2024] (I) White balance...
[19:27:37 - 17/10/2024] (I) Saving /tmp/.satnogs/satdump_10416022/raw_unsync.png...
[19:27:37 - 17/10/2024] (T) Tried to save empty PNG!
[19:27:37 - 17/10/2024] (I) Synchronize...
[19:27:37 - 17/10/2024] (T) Wedge 1
[19:27:37 - 17/10/2024] (T) Wedge 2
[19:27:37 - 17/10/2024] (C) Couldn't calibrate image with wedges! Likely no valid wedge was identified!
[19:27:37 - 17/10/2024] (I) Saving /tmp/.satnogs/satdump_10416022/raw_sync.png...
[19:27:37 - 17/10/2024] (T) Tried to save empty PNG!
[19:27:37 - 17/10/2024] (I) ----------- Satellite
[19:27:37 - 17/10/2024] (I) NORAD : 33591
[19:27:37 - 17/10/2024] (I) Name  : NOAA-19
[19:27:37 - 17/10/2024] (I) ----------- AVHRR
[19:27:37 - 17/10/2024] (I) Lines : 0
[19:27:37 - 17/10/2024] (W) Identified channels (debug): A: -1 A1: -1 B: -1
[19:27:37 - 17/10/2024] (I) No valid timing data; not aligning timestamp
[19:27:37 - 17/10/2024] (I) Saving /tmp/.satnogs/satdump_10416022/APT-A.png...
[19:27:37 - 17/10/2024] (T) Tried to save empty PNG!
[19:27:37 - 17/10/2024] (I) Saving /tmp/.satnogs/satdump_10416022/APT-B.png...
[19:27:37 - 17/10/2024] (T) Tried to save empty PNG!
[19:27:37 - 17/10/2024] (I) Done! Goodbye

f4tnk, 2024-10-18:
I'm going to check my udp source... I may have a problem at this level even if I think it is active

f4tnk, 2024-10-18:
UDP Source is OK and decode via IQ source file is OK... any idea why net_source not work ?

JVital2013, 2024-10-19:
Hello! Does it work when you try it from the UI?

I believe you need to specify the mode with the current version, like this:

```bash
satdump live noaa_apt $OUT --source net_source --mode udp --port $UDP_DUMP_PORT --satellite_number 15 --samplerate $SAMP --frequency $FREQ
```

f4tnk, 2024-10-19:
Hello,

It's okay with -mode udp :-)

My first idea with the code was therefore the right one...it was simple finally...

I will go to Matrix next time rather than opening an issue.

Thanks for the fantastic work you do.

closed 2024-10-19 by f4tnk