SatDump/SatDump #840: SatDump crashes when processing APT wav

mathuin, 2024-10-22:
**Description of the issue**
I have a WAV file I recorded from a NOAA 18 pass that I tried to process with SatDump, and SatDump crashed.

**Hardware (SDR/PC/OS)**
Raspberry Pi 4 running debian bullseye, file was recorded with fldigi from an IC-7100 with a mag mount 2m/440 antenna.

**Version (Eg, 1.0.0, CI Build 171)**
v1.2.1-323d2c0b

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
[satdump-20241021T162711.log](https://github.com/user-attachments/files/17467978/satdump-20241021T162711.log)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
The WAV is too big to attach here.

mathuin, 2024-10-22:
I am able to reproduce the crash on a MacBook Pro with an M1 Max chip -- it's a lot faster, but still crashes.

JVital2013, 2024-11-18:
Hi @mathuin,

Sorry for the late reply - I believe I fixed this issue towards the beginning of November. Can you update and try decoding your wav again?

If it still doesn't work, we will need a copy of the wav. Just upload it somewhere else, like SwissTransfer or Google Drive.

mpentler, 2024-11-22:
Ping @mathuin - how are you getting on with this?

mathuin, 2024-11-26:
I was able to work on this today.  I updated my Raspberry Pi's git checkout to the current commit (appeasing penguin overlords) and rebuilt the software.  It's now v1.2.2-ec87d43c.

I still get the same error.  Here's the WAV file: https://www.swisstransfer.com/d/f4d865f1-c95c-46f2-b853-5646efe4238b

mathuin, 2024-11-26:
I also tried the nightly build for macOS, and it also failed the same way it did earlier (same message, but did not actually crash)

JVital2013, 2024-11-27:
Thanks for the file! I'm glad to hear the crash is fixed, and you just get an error now.

The problem is that your signal is noisy, and I'm suspicious either the full bandwidth of the signal was not captured, or the signal was compressed at some point. You can see here that these "sync lines" are unfortunately indistinguishable from the background noise, even at some of the cleaner areas:

[image]

There's not enough good data to pull any information out. The errors in the log occur because this data is missing, which is to be expected. You do get some nice clouds in the unsynced image though!

[image]

closed 2024-11-28 by JVital2013