SatDump/SatDump #843: FFT range (Waterfall/Spectrum) does not fit for Airspy HF+ Discovery - outside of display

db2os, 2024-10-26:
No Spectrum/Waterfall with Airspy HF+ Discovery SDR because FFT Min setting is out of range (-150 limit).

- Hardware (AirspyHF, Win11 64 bit)
- latest nightly

below some screenshots in comparison with SDR++,  not that the Minimum slider is at maximum too.

[image]
[image]

JVital2013, 2024-10-27:
Yes, this is something we need to address. For now, you can set the FFT min to a number lower than -150 by Ctrl+Clicking on the FFT Min slider, and typing a lower number (such as -180)

db2os, 2024-10-27:
> Yes, this is something we need to address. For now, you can set the FFT min to a number lower than -150 by Ctrl+Clicking on the FFT Min slider, and typing a lower number (such as -180)

Oh,  that's a trick I did'nt know!!  Works nice..   Thanks for the hint!

JVital2013, 2024-11-21:
Hey @db2os, what would be a good minimum value for use with the Airspy HF+ Discovery? How low should we go?

db2os, 2024-11-23:
I think -160 should be fine

JVital2013, 2024-11-24:
I'll just set that as a new minimum. We had talked about making the FFT range more dynamic, but this should suffice for now.

commit dd55a27d (2024-11-24): Fix #843
closed 2024-11-24 by JVital2013