SatDump/SatDump #844: Autotrack focuses on one satellite

super-15, 2024-10-27:
**Description of the issue**
After recieving one satellite using autotrack, it will wait to recieve the next pass of that satellite, not go to the next, nearest satellite.
**Hardware (SDR/PC/OS)**
SDRPlay RSPdx, Windows 10 freshly installed
**Version (Eg, 1.0.0, CI Build 171)**
Latest nightly build as of 10/26/2024

JVital2013, 2024-10-27:
Could you please share a log and your `config.json` of this happening? I am unable top reproduce this problem - it always switches to the next satellite at its AOS

JC167shearer, 2024-11-01:
I have this issue, too:

Actively tracking:

[image]
[image]

After that pass, scheduler shows NOAA15 next, but Object Tracker still focused on Meteor M2-4:

[image]
[image]

Will get json and logs this evening if the OP doesn't get it first. Thank you!

JVital2013, 2024-11-01:
Thanks for the additional report. To clarify: does it actually start the processing for the correct satellite at AOS, and only the web interface is wrong? So far that looks like it might be the actual issue - the web interface is switching, but the tracking itself may still be OK

super-15, 2024-11-01:
Yes, it seems only an issue with the UI. It switches at AOS.

JC167shearer, 2024-11-01:
Correct. The UI stays "focused" (for lack of a better term) on the sat it just completed rather than shifting focus to the next sat on the scheduler.

JVital2013, 2024-11-01:
OK - we will troubleshoot. Thanks all!

closed 2024-11-03 by JVital2013

JVital2013, 2024-11-03:
This should now be fixed in the latest nightly build. Let us know if you still have issues!

commit c9f98bff (2024-11-03): Fix #844