SatDump/SatDump #847: Source type: remote no handler

burgeda, 2024-11-01:
I updated satdump from git on Friday Nov 1st  and now I get 

[21:22:45 - 01/11/2024] (E) Could not find a handler for source type : remote!

in the log file, satdump core dumps, starts again ... rinse and repeat.

I'm running satdump_sdr_server on a pi and satdump on a dell micro, this is the service file that has been working fine for more than 6 months.

/usr/bin/satdump live goes_hrit /srv/sat-photos --source remote --source_id 00000001 --samplerate 2.4e6 --frequency 1694.1e6 --gain 30 --agc true --http_server 0.0.0.0:8080

JVital2013, 2024-11-01:
Hello! Please share your entire log of this happening so we can troubleshoot. It sounds like the remote sdr plugin is not loading

burgeda, 2024-11-02:
[satdump-20241101T212244.log](https://github.com/user-attachments/files/17605104/satdump-20241101T212244.log)

burgeda, 2024-11-02:
Well, that's that one solved, there was an error in the settings.json file and it wasn't loading.
What's happening now is that it runs for about a minute, core dumps and starts again.

[satdump-20241102T012806.log](https://github.com/user-attachments/files/17605129/satdump-20241102T012806.log)

burgeda, 2024-11-02:
I decided to fire up satdump-ui after many many months of just using the command line satdump and this is what I'm getting:

satdump-ui: Symbol `_ZN5style5themeE' has different size in shared object, consider re-linking
satdump-ui: symbol lookup error: satdump-ui: undefined symbol: re_compile

I've recompiled and reboot many times and the result is always the same.

burgeda, 2024-11-02:
Fixed.
Thinking it might have something to do with the library fixes for linux
I did:

git cloned the latest
recompiled it
make install
make uninstall

Then I search the entire file system for any "satd" files and directories.
Manually deleted everything that wasn't in my build directory

did make install

It not starts up.  

SatDump v1.2.2-da5f25eb

JVital2013, 2024-11-02:
Yeah, you beat me to finding the solution! There was a change in the last month or so that moved where the libraries live on many systems. If you built from source and did not run "sudo make uninstall" before cloning/pulling the latest - it would leave old files behind, and SatDump would sometimes load them instead.

People using the pre-built packages, or uninstalling before pulling the latest code, do not have the problem. The fix is to manually delete the old files as you noted. We also have a bit about this in the 1.2.1 changelog.

closed 2024-11-02 by JVital2013