SatDump/SatDump #852: Continous Coredump

burgeda, 2024-11-02:
SatDump v1.2.2-da5f25eb continuously coredumps.

                                         Module libSPIRV-Tools.so from rpm spirv-tools-2024.3-3.fc40.x86_64
                                                  Module libigc.so.1 from rpm intel-igc-1.0.17224.11-1.fc40.x86_64
                                                  Module libtinfo.so.6 from rpm ncurses-6.4-12.20240127.fc40.x86_64
                                                  Module libedit.so.0 from rpm libedit-3.1-53.20240808cvs.fc40.x86_64
                                                  Module libffi.so.8 from rpm libffi-3.4.4-7.fc40.x86_64
                                                  Module libLLVMSPIRVLib.so.15 from rpm spirv-llvm15.0-translator-15.0.0-9.fc40.x86_64
                                                  Module libopencl-clang.so.15 from rpm intel-opencl-clang-15.0.0-11.fc40.x86_64
                                                  Module libigdfcl.so.1 from rpm intel-igc-1.0.17224.11-1.fc40.x86_64
                                                  Module libigdgmm.so.12 from rpm intel-gmmlib-22.3.20-1.fc40.x86_64
                                                  Module libigdrcl.so from rpm intel-compute-runtime-24.26.30049.6-1.fc40.x86_64
                                                  Module libhackrf.so.0 from rpm hackrf-2024.02.1-1.fc40.x86_64
                                                  Module libsz.so.2 from rpm libaec-1.1.2-1.fc40.x86_64
                                                  Module libhdf5_hl.so.200 from rpm hdf5-1.12.1-15.fc40.x86_64
                                                  Module libhdf5.so.200 from rpm hdf5-1.12.1-15.fc40.x86_64
                                                  Module libcap.so.2 from rpm libcap-2.69-8.fc40.x86_64
                                                  Module libudev.so.1 from rpm systemd-255.13-1.fc40.x86_64
                                                  Module libusb-1.0.so.0 from rpm libusb1-1.0.27-2.fc40.x86_64
                                                  Module libairspy.so.0 from rpm airspyone_host-1.0.10-6.fc40.x86_64
                                                  Module libsamplerate.so.0 from rpm libsamplerate-0.2.2-8.fc40.x86_64
                                                  Module libopus.so.0 from rpm opus-1.5.1-1.fc40.x86_64
                                                  Module libjack.so.0 from rpm jack-audio-connection-kit-1.9.22-5.fc40.x86_64
                                                  Module libasound.so.2 from rpm alsa-lib-1.2.12-1.fc40.x86_64
                                                  Module libportaudio.so.2 from rpm portaudio-19-43.fc40.x86_64
                                                  Module libopenblaso.so.0 from rpm openblas-0.3.26-4.fc40.x86_64
                                                  Module libflexiblas_openblas-openmp.so from rpm flexiblas-3.4.4-1.fc40.x86_64
                                                  Module libflexiblas_fallback_lapack.so from rpm flexiblas-3.4.4-1.fc40.x86_64
                                                  Module libflexiblas_netlib.so from rpm flexiblas-3.4.4-1.fc40.x86_64
                                                  Module libpcre2-8.so.0 from rpm pcre2-10.44-1.fc40.x86_64
                                                  Module libcrypt.so.2 from rpm libxcrypt-4.4.36-5.fc40.x86_64
                                                  Module libselinux.so.1 from rpm libselinux-3.7-5.fc40.x86_64
                                                  Module libpcre2-posix.so.3 from rpm pcre2-10.44-1.fc40.x86_64
                                                  Module libsharpyuv.so.0 from rpm libwebp-1.3.2-5.fc40.x86_64
                                                  Module libbrotlicommon.so.1 from rpm brotli-1.1.0-3.fc40.x86_64
                                                  Module libsasl2.so.3 from rpm cyrus-sasl-2.1.28-19.fc40.x86_64
                                                  Module libevent-2.1.so.7 from rpm libevent-2.1.12-12.fc40.x86_64
                                                  Module libkeyutils.so.1 from rpm keyutils-1.6.3-3.fc40.x86_64
                                                  Module libkrb5support.so.0 from rpm krb5-1.21.3-1.fc40.x86_64
                                                  Module libcom_err.so.2 from rpm e2fsprogs-1.47.0-5.fc40.x86_64
                                                  Module libk5crypto.so.3 from rpm krb5-1.21.3-1.fc40.x86_64
                                                  Module libkrb5.so.3 from rpm krb5-1.21.3-1.fc40.x86_64
                                                  Module libunistring.so.5 from rpm libunistring-1.1-7.fc40.x86_64
                                                  Module libmetis.so.0 from rpm metis-5.1.0.3-5.fc40.x86_64
                                                  Module libjpeg.so.62 from rpm libjpeg-turbo-3.0.2-1.fc40.x86_64
                                                  Module libjbig.so.2.1 from rpm jbigkit-2.1-29.fc40.x86_64
                                                  Module libLerc.so.4 from rpm liblerc-4.0.0-6.fc40.x86_64
                                                  Module libwebp.so.7 from rpm libwebp-1.3.2-5.fc40.x86_64
                                                  Module libbrotlidec.so.1 from rpm brotli-1.1.0-3.fc40.x86_64
                                                  Module libgssapi_krb5.so.2 from rpm krb5-1.21.3-1.fc40.x86_64
                                                  Module libcrypto.so.3 from rpm openssl-3.2.2-3.fc40.x86_64
                                                  Module libssl.so.3 from rpm openssl-3.2.2-3.fc40.x86_64
                                                  Module libpsl.so.5 from rpm libpsl-0.21.5-3.fc40.x86_64
                                                  Module libssh.so.4 from rpm libssh-0.10.6-5.fc40.x86_64
                                                  Module libidn2.so.0 from rpm libidn2-2.3.7-1.fc40.x86_64
                                                  Module libnghttp2.so.14 from rpm nghttp2-1.59.0-3.fc40.x86_64
                                                  Module libmbedx509.so.1 from rpm mbedtls-2.28.8-1.fc40.x86_64
                                                  Module libmbedcrypto.so.7 from rpm mbedtls-2.28.8-1.fc40.x86_64
                                                  Module libmbedtls.so.14 from rpm mbedtls-2.28.8-1.fc40.x86_64
                                                  Module liborc-0.4.so.0 from rpm orc-0.4.39-1.fc40.x86_64
                                                  Module libcpu_features.so.0 from rpm google-cpu_features-0.7.0-6.fc40.x86_64
                                                  Module libsuperlu.so.6 from rpm SuperLU-6.0.1-3.fc40.x86_64
                                                  Module libarpack.so.2 from rpm arpack-3.9.1-3.fc40.x86_64
                                                  Module libflexiblas.so.3 from rpm flexiblas-3.4.4-1.fc40.x86_64
                                                  Module libtiff.so.6 from rpm libtiff-4.6.0-5.fc40.1.x86_64
                                                  Module libcurl.so.4 from rpm curl-8.6.0-10.fc40.x86_64
                                                  Module libzstd.so.1 from rpm zstd-1.5.6-1.fc40.x86_64
                                                  Module libnng.so.1 from rpm nng-1.7.2-1.fc40.x86_64
                                                  Module libfftw3f.so.3 from rpm fftw-3.3.10-12.fc40.x86_64
                                                  Module libz.so.1 from rpm zlib-ng-2.1.7-2.fc40.x86_64
[... 17 lines cut ...]
                                                  #11 0x00007fe4841f371d _ZN3dsp23getSourceFromDescriptorENS_16SourceDescriptorE (libsatdump_core.so + 0x1f371d)
                                                  #12 0x0000000000452bd2 _Z9main_liveiPPc (satdump + 0x52bd2)
                                                  #13 0x000000000041b832 main (satdump + 0x1b832)
                                                  #14 0x00007fe482e39088 __libc_start_call_main (libc.so.6 + 0x2a088)
                                                  #15 0x00007fe482e3914b __libc_start_main@@GLIBC_2.34 (libc.so.6 + 0x2a14b)
                                                  #16 0x000000000041bb25 _start (satdump + 0x1bb25)
                                                  
[... cut ...]
                                                  #1  0x00007fe45caebf93 linux_udev_event_thread_main (libusb-1.0.so.0 + 0x10f93)
                                                  #2  0x00007fe482ea66d7 start_thread (libc.so.6 + 0x976d7)
                                                  #3  0x00007fe482f2a60c __clone3 (libc.so.6 + 0x11b60c)
                                                  ELF object binary architecture: AMD x86-64

I've attached the latest coredump file generated by systemd

[Uploading core.satdump.1000.1df940d885d547a4b0b0352d6b608c71.113544.1730554254000000.zst.gz…]()

burgeda, 2024-11-02:
It seems to write one GOES-16 photo and then crash, doesn't log much 
[satdump-20241102T143133.log](https://github.com/user-attachments/files/17607289/satdump-20241102T143133.log)

JVital2013, 2024-11-02:
Have you modified your composites in any way in settings.json, to override composite settings in satdump_cfg.json? If so, double check that you've updated the settings for the new composite names in 1.2.1/1.2.2.

I do intend to get that crash fixed though, if that's what you're running into

burgeda, 2024-11-02:
Thanks for the quick response.
Yes, I had indeed overridden the composite settings - added autogen true to all of them.
I removed my settings.yml and recreated it using the GUI.
All seems well now.  Or at least the software is running now I need to wrap my head around the changes and neat new features.

It's received and written a bunch of GOES photos without issue so ... problem solved.

One other thing I noted is that even though I'm running the GUI on a box without an rtlsdr the software now insists that I have the libraries for librtlsdr installed.  Didn't before.  Minor issue that even I could spot.  Just thought I'd mention it.

Thanks again for getting me up and running.

JVital2013, 2024-11-02:
No problem! I just fixed the nightlies so it won't crash for other people that end up in your situation.

I'm curious, where are you seeing it requiring librtlsdr? The .deb package has been requiring it for about a year or so to make it easier for beginners to get everything they need at once, but building from source should allow you to build without it

burgeda, 2024-11-03:
It was right in the log shortly after startup, it complained that it couldn't load /usr/lib64/satdump/librtlstr_sdr_support.so and then it crashed.
However it's possible that the crash was just a coincedence as it was core dumping repeatedly at that time.

I did a find on the file system and there was no librtl* of any kind, not surprising as I had never installed it so I downloaded it (rtlsdr) from github, maked it, installed it, did a ldconfig and after that satdump-ui stopped complaining about it.  Oddly I had never noticed an entry in the log like that when running just the cli satdump.

Could just be a red herring as things seem to be working again, I have to tweak some of the vitality-goes config files as not all of the GOES photos are showing up but that shouldn't be too much of a  problem (famous last words).

---- On Sat, 02 Nov 2024 16:25:42 -0400 Jamie Vital ***@***.***> wrote ---

No problem! I just fixed the nightlies so it won't crash for other people that end up in your situation.

I'm curious, where are you seeing it requiring librtlsdr? The .deb package has been requiring it for about a year or so to make it easier for beginners to get everything they need at once, but building from source should allow you to build without it

JVital2013, 2024-11-03:
Yeah, it was likely a red herring. We've also recently (in the last 6 months) improved some of the messages when plugins fail to load. It was probably failing before, just less noticeably (and silently).

As for Vitality GOES, most of the necessary changes should be visible here: https://github.com/JVital2013/vitality-goes/commit/d720b6da3765bae86a97cc8609f45e87737dcb21.

closed 2024-11-03 by JVital2013

burgeda, 2024-11-03:
Thanks for that link, I am literally in there this second making those changes.

---- On Sat, 02 Nov 2024 21:21:44 -0400 Jamie Vital ***@***.***> wrote ---

Yeah, it was likely a red herring. We've also recently (in the last 6 months) improved some of the messages when plugins fail to load. It was probably failing before, just less noticeably (and silently).

As for Vitality GOES, most of the necessary changes should be visible here: https://github.com/JVital2013/vitality-goes/commit/d720b6da3765bae86a97cc8609f45e87737dcb21.