SatDump/SatDump #856: Could not open RTL-SDR device!

heliumn55, 2024-11-05:
Hello, 

using last release with windows 11 24H2, with an original  RTL-SRD.COM V4.
 Satdump works well with my sdrplay rspduo.

but with V4 nothing.
it says :  Could not open RTL-SDR device!   look picture below

[image]

very good work for that freeware

JVital2013, 2024-11-05:
Based on the screenshot, you have not installed the driver correctly with Zadig (OEM #0 is the tell-tale sign here). Make sure you've properly installed the RTL-SDR driver with Zadig and try again.

These instructions still appear to be valid: https://www.rtl-sdr.com/getting-the-rtl-sdr-to-work-on-windows-10/

heliumn55, 2024-11-05:
My V4 is already configured with Zadig for SDRSHARP!
it works very well

heliumn55, 2024-11-05:
Ok You're right.
reinstall drivers again with zadig 2.9.
now it works

Thank you

closed 2024-11-05 by heliumn55