SatDump/SatDump #858: USRP performance and indication issue, v1.2.1

O-I812, 2024-11-08:
Hello! First, thank you for SatDump, it is an incredible piece of decoding software, and makes SDR's a lot if fun! I recently updated SatDump to v1.2.1 and found a few issues compared to the last version. I am using two different SDR's ,one is a clone LibreSDR B210 with the new FPGA chip set, however I have also seen the same issue using different binaries with a original Ettus B210. This issues are surrounding performance and what is indicated in the GUI. 

1) Under the 'recoder' menu, I can select the USRP B210 and serial number, and everything works as normal. When I select a sample rate of 2 Msps things work normally, however when I increase the sample rate (8-bit mode on the USRP) the FFT and waterfall get very slow and lag by a few seconds. I use SatDump for Aura and Aqua passes, so my sample rate needs to be around 30 Msps. Even though this challenges the CPU, it seemed to work just fine on previous versions of SatDump. I have not tried a NOAA pass with the new version, so I am not sure if this affects the demod or frame sync process yet. I have tried to set the 'OpenCL Device' to use the CPU to see if that helped or degraded performance, but there was no change. 

2) The other issue I see was when using the USRP, and selecting the correct sample rate (e.g. 4 Msps) when I hit 'start' the sample rate indication always defaults back to 1 Msps. I have however confirmed that the sample rate is set properly with the USRP as the FFT shows the correct bandwidth, it just seems to be an issue with the GUI indication. 
  
Hardware: LibreSDR B210 clone (libre bins), Ettus B210 (ettus bins), Minisforum Mini PC NAB9 (i9-12900HK 14 cores; 32 GB RAM; 1TB)

Version: 1.2.1. I did read the release notes to see if there was any update to the sampling of the FFT to make it slower, however I did not find any indication of that.
 
- No crashes to report, just slow performance. 

Again, thank you, it is amazing software!

Robydem, 2024-11-09:
Hi everyone, same problem with B200!!

Robydem, 2024-11-10:
However, I want to point out that in my case, it is not just a display problem; the recording is 1 msps and therefore unusable.

Robydem, 2024-11-18:
> However, I want to point out that in my case, it is not just a display problem; the recording is 1 msps and therefore unusable.

Hi all,
for now I solved it by installing version 1.1.4 which works perfectly in BB recording. All the "overflow errors" have also disappeared (even over 50 msps). At the same time I also installed the latest version which I use only for decoding.

O-I812, 2024-11-19:
May seem like a dumb question, but I have not had to revert to an earlier version of SatDump, does anyone know how to go about doing that on Ubuntu 22.04?  I tried going to the 'releases' page, but version 1.1.4 tells me about the features, but doesn't have an installer. Is there somewhere I can go for archived installers? Any help would be appreciated!

Robydem, 2024-11-19:
Go here:  https://github.com/SatDump/SatDump/releases/tag/1.1.4

and get Source code .zip and unzip it into a SatDump-1.1.4 folder and put it in the same location where you have the SatDump folder (Home). Then open the terminal and install:

cd SatDump-1.1.4
mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr ..
make -j`nproc`

sudo make install

./satdump-ui

Note: you have to set the writing folders and all the parameters in the menu of the latest version and then you will find them also in the old version.

ciao
Roberto IW0EHE

O-I812, 2024-11-20:
Thank you Roberto! I appreciate the guidance, I will give that a try! Not a Linux pro either, but I make it work! Thanks again!

dengrp, 2025-06-10:
Still occurring in version 1.2.2 with B200. Trying to use SatDump with GOES 19 GRB and IS-21 GEONETcast. No matter what Sample rate is set to, once "Run" is selected, the sample rate is set by SatDump to 1MSps, so cannot operate with GOES 19 or IS-21 GEONETcast.

dengrp, 2025-06-21:
Additional information. With the Ettus USRP B200 and version 1.2.2 of SatDump under Windows 10 Pro O/S, after selecting any sample rate above 1Msps and then selecting 'START', the bandwidth shown on the Waterfall display is correct; however, the sample rate indicated drops back to 1Msps and the actual sample rate of the B200 is set to 1 Msps. I am working on reception of GOES 18 and 19 GRB as well as GEONETcast, so need sample rates of 16 Msps and 30 Msps. I have all the hardware ready, am receiving GOES GRB with SNR of 12.8 dB and GEONETcast with SNR of 11.8 dB, and am glad to test, if corrections can be made.

sergsero, 2025-07-27:
Unfortunately, the issues with the Ettus USRP B200/B210 that were described above ("drops back to 1Msps" and "FFT and waterfall get very slow") still exists in version 1.2.3 and the new verywip branch (v.2.x).

Patrick-Cunningham, 2025-07-27:
A pull request I have up should resolve this! https://github.com/SatDump/SatDump/pull/1007

sergsero, 2025-07-28:
Thank you very much for finding the issue solution and making the necessary changes to the code so quickly!

Robydem, 2025-07-28:
Thank you very much !! Now it works perfectly