SatDump/SatDump #860: Crash upon loading .wav file into UI since upgrade to latest version (Windows and Mac OS) .

CharlesPlusC, 2024-11-09:
**Description of the issue**
Since I upgraded to the latest build of sat dump I have the following issue. As soon as I load my .wav file into sat dump it just closes down. I go click the open file next to "Input File", then select my .wav file, click "choose" and then the whole thing just closes down. I usually work on MacOS but also tried this on my friends PC and the same thing occurred. I'm not sure if I recorded this NOAA APT pass incorrectly or something... Usually works fine. Thanks for any help 😃 📡 

**Hardware (SDR/PC/OS)**
Mac OS, Apple Silicon, Sonomoa 14.6.1

**Version (Eg, 1.0.0, CI Build 171)**
Running the latest available build of SatDump. v1.2.1

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

[satdump-20241109T214548.log](https://github.com/user-attachments/files/17689505/satdump-20241109T214548.log)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

JVital2013, 2024-11-17:
Hello!

Would it be possible to share the wav here? What program did it come from? I'm wondering if SatDump is choking on something specific in the wav.

CharlesPlusC, 2024-11-22:
It seems I can't attach .wav files here... Also it is very large (10.14GB). Perhaps I just recorded it wrong... Will try again at some point this weekend to see if I can replicate. 🤔

JVital2013, 2024-11-22:
At that size it sounds like a baseband wav - again, what program did it come from? Maybe we can reproduce if we know the source program.

Either way, sounds good about this weekend!

CharlesPlusC, 2024-11-22:
It was from SDR++. Will keep you posted 😃

JVital2013, 2024-11-28:
I took some time and was able to reproduce your issue. It is now fixed as of the latest nightly. Thanks for the report!

closed 2024-11-28 by JVital2013

CharlesPlusC, 2024-11-28:
Hi,
Apologies for not having been able to replicate the issue for you once more... Thanks so much for fixing this 🙏 😄  looking forward to getting back up and running.