SatDump/SatDump #864: Satdump on Pi error

hjerald1, 2024-11-15:
Rasp Pi 4 latest Bookworm 64 bit w/GUI. Satdump 1.2.1 installed from website link, [64-Bit ARM (like Raspberry Pi OS; deb)].

Satdump UI opens. Selected Generic rtl-sdr (using either rtl-sdr v4 or nooelec smarteextr), hit start and this error appears:   plugins/sdr_sources/rtlsdr_sdr_support/rtlsdr_sdr.cpp:214

Cmd "rtl_test" successfull using either sdr.

Can someone help with this error?
 I'm rather a noob with Linux and Rasp Pi.

hjerald1, 2024-11-15:
So in the meantime, I started over and tried to compile satdump myself. All went well until the make -j`nproc` command. Things proceeded until 45% then froze.
So I started over with a fresh install of the latest 64 bit Bookworm with desktop. Loaded nightly build of [Debian 64-Bit ARM (like Raspberry Pi OS). Then tried installation with   sudo apt install ./satdump_rpi64_latest_arm64.deb   . 
Got the the message:     
Reading package lists... Done
E: Unsupported file ./satdump_rpi64_latest_arm64.deb given on commandline

I'm out of ideas.......

krakenrf, 2024-11-17:
Instead of make -j nproc, try make -j 3. I also had trouble with SatDump compilation crashing the entire Pi when nproc is used. Using one core less helps.

hjerald1, 2024-11-17:
Thanks, will give it a try.