SatDump/SatDump #866: Crash when saving .png

NicolasGagne, 2024-11-16:
**Description of the issue**
Satdump crash repetely when it save .png file. 
I run satdump with the tracking mode and track NOAA and Meteor at 137Mhz.
The error repete itself regularly pretty much each 2 passé of NOAA satellite

**Hardware (SDR/PC/OS)**
Ubuntu 20.04, I5-4XXX Cpu.

**Version (Eg, 1.0.0, CI Build 171)**
v1.2.2-4b25fccd build form source 
I try uninstall and rebuild from source and same issue. 

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
[satdump-20241115T223420.log](https://github.com/user-attachments/files/17787884/satdump-20241115T223420.log)
If I run Satdump in the terminal the last line I get is Abort(core dump)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

JVital2013, 2024-11-17:
Hello again!

Do you have any familiarity with gdb? Is there any chance you could rebuild in Debug or DebWithRelInfo, then try doing the same thing while running in gdb? When the crash happens, type `bt` and it should help narrow down the issue.

NicolasGagne, 2024-11-17:
Thanks JiVital2013 for the suggestion but I believe I found the problem

The way Satdump autotrack works
- it create the folder for each pass when the pass start  
- when the pass ends it look for that folder created at the beginning to save the differents files and create the dataset.json file. 

The problem was that my webserver was deleting the folder when it wasn't seeing the dataset.json file in the folder. 
I have modified my webserver to account for this problem and no more issue.

If you are interested: https://github.com/NicolasGagne/Satdump_webpage

thanks.

closed 2024-11-17 by NicolasGagne

JVital2013, 2024-11-18:
I'm glad you worked it out, and thanks for sharing your site!