SatDump/SatDump #867: SatDump randomly killed on Pi 4

krakenrf, 2024-11-17:
SatDump keeps crashing with a 'Killed' error. Currently, I am running on a Pi 4 2GB, latest 64-bit Raspbian.

My setup has SatDump running an RTL-SDR at 2.56 MSPS, rotator tracking, and Schedule and Config enabled, and currently tracking NOAA 18, 19, METEOR-M2 3 and METEOR M2 4.

It can sometimes run fine for a couple of passes, but then another pass will just crash SatDump. The only error in the console is 'Killed'.

In the settings Open Viewer after decoding is turned off. All other settings are default.

```
[11:40:36 - 16/11/2024] (I) ----------- Satellite
[11:40:36 - 16/11/2024] (I) NORAD : 28654
[11:40:36 - 16/11/2024] (I) Name  : NOAA-18
[11:40:36 - 16/11/2024] (W) NOAA-18 detected. MHS data will not be saved!
[11:40:36 - 16/11/2024] (I) ----------- AVHRR/3
[11:40:36 - 16/11/2024] (I) Lines : 5054
[11:40:38 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AVHRR/AVHRR-5.png...
[11:40:38 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AVHRR/AVHRR-4.png...
[11:40:38 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AVHRR/AVHRR-3a.png...
[11:40:38 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AVHRR/AVHRR-1.png...
[11:40:38 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AVHRR/AVHRR-3b.png...
[11:40:47 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AVHRR/AVHRR-2.png...
[11:40:58 - 16/11/2024] (I) ----------- HIRS
[11:40:58 - 16/11/2024] (I) Lines : 131
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-16.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-6.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-7.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-11.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-8.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-1.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-2.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-17.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-12.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-13.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-18.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-19.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-14.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-3.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-9.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-20.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-10.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-15.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-4.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/HIRS/HIRS-5.png...
[11:40:58 - 16/11/2024] (I) ----------- SEM
[11:40:58 - 16/11/2024] (I) Sample counts from selected channels :
[11:40:58 - 16/11/2024] (I) Channel OP1   : 412
[11:40:58 - 16/11/2024] (I) Channel P8    : 205
[11:40:58 - 16/11/2024] (I) Channel 0DE1  : 77
[11:40:58 - 16/11/2024] (I) Channel 0EFL  : 23
[11:40:58 - 16/11/2024] (I) Backgrounds   : 27
[11:40:58 - 16/11/2024] (I) ----------- AMSU
[11:40:58 - 16/11/2024] (I) Lines A1: 96
[11:40:58 - 16/11/2024] (I) Lines A2: 98
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AMSU/AMSU-A-5.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AMSU/AMSU-A-6.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AMSU/AMSU-A-13.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AMSU/AMSU-A-7.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AMSU/AMSU-A-14.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AMSU/AMSU-A-1.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AMSU/AMSU-A-9.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AMSU/AMSU-A-2.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AMSU/AMSU-A-15.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AMSU/AMSU-A-8.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AMSU/AMSU-A-10.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AMSU/AMSU-A-3.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AMSU/AMSU-A-4.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AMSU/AMSU-A-11.png...
[11:40:58 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AMSU/AMSU-A-12.png...
[11:40:58 - 16/11/2024] (D) Products processing is enabled! Starting processing module.
[11:40:58 - 16/11/2024] (I) ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AVHRR
[11:40:58 - 16/11/2024] (I) ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AVHRR/product.cbor
[11:40:58 - 16/11/2024] (I) ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AVHRR/product.cbor
[11:40:59 - 16/11/2024] (I) Loading AVHRR-1.png
[11:40:59 - 16/11/2024] (I) Loading AVHRR-2.png
[11:40:59 - 16/11/2024] (I) Loading AVHRR-3a.png
[11:41:00 - 16/11/2024] (I) Loading AVHRR-3b.png
[11:41:00 - 16/11/2024] (I) Loading AVHRR-4.png
[11:41:01 - 16/11/2024] (I) Loading AVHRR-5.png
[11:41:02 - 16/11/2024] (D) Composite needs channel ch1
[11:41:02 - 16/11/2024] (D) Composite needs channel ch2
[11:41:02 - 16/11/2024] (T) Max Total Width 2048
[11:41:02 - 16/11/2024] (T) Max Total Used  2048
[11:41:02 - 16/11/2024] (T) Offset for ch1 is 0
[11:41:02 - 16/11/2024] (T) Offset for ch2 is 0
[11:41:06 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AVHRR/avhrr_3_rgb_AVHRR_221_False_Color.png...
[11:41:15 - 16/11/2024] (I) Saving ./live_output/2024-11-16_11-24_noaa_hrpt_1.707 GHz/AVHRR/avhrr_3_rgb_AVHRR_221_False_Color_corrected.png...
[11:41:31 - 16/11/2024] (D) Reprojecting composite avhrr_3_rgb_AVHRR_221_False_Color
[11:41:31 - 16/11/2024] (D) Filtering timestamps...
[11:41:31 - 16/11/2024] (T) Warping size 20480x10240
[11:41:32 - 16/11/2024] (T) Smart Warp using 20480 2958, full size 20480 10240
[11:41:36 - 16/11/2024] (T) We will split into 1 segments. Median distance is 108.657655 km and total (avg) distance is 5432.882740 km
[11:41:36 - 16/11/2024] (I) Solving TPS equations for 1017 GCPs...
[11:41:42 - 16/11/2024] (I) Solved! Took 6.029173
Killed

```

JVital2013, 2024-11-17:
Hello!

Usually this means the Linux kernel killed the process for some reason. I'm guessing we're running into a memory issue. Can you run `sudo journalctl --since "1 hour ago"` (update the time as necessary) and see if there's anything in the system log at the same time SatDump was killed? If it's a memory issue, you should see something about an `oom-killer`.

It's been a while since I tested HRPT on a system with only 2GB. It may be worth looking again to see if too much memory is being allocated somewhere.

JVital2013, 2024-11-18:
Hi @krakenrf,

Do you mind updating and trying again? I went through and audited memory usage with the way you're using SatDump, and was able to make some significant savings (almost 300 MB!) I'm guessing the problem you're having is from hitting the memory limit of your Pi, and my changes should help.

If it still crashes after updating, can you share the .frm/.cadu that it got killed on?

krakenrf, 2024-11-21:
Thanks, I will try again soon. Just took the setup down temporarily, should have it back up in a week or so.