SatDump/SatDump #872: Error appearing when build  SDDC plugin

crackerjacques, 2024-11-22:
When build with SDDC SDR UNIT support on Ubuntu 24.04

In

**sddc_sdr.h**

` 7 #include "common/dsp/rational_resampler.h"`
↓fix↓
` 7 #include "common/dsp/resamp/rational_resampler.h"`

and **sddc_sdr.cpp**

/home/jackheinlein/test/SatDump/plugins/sdr_sources/sddc_sdr_support/sddc_sdr.cpp:68:34: error: cannot convert ‘std::string’ {aka ‘std::__cxx11::basic_string<char>’} to ‘int’

`68 sddc_dev_obj = sddc_open(d_sdr_id, resources::getResourcePath("sddc/SDDC_FX3.img").c_str());`

↑Error on this line.
↓fix
`68 sddc_dev_obj = sddc_open(std::stoi(d_sdr_id), resources::getResourcePath("sddc/SDDC_FX3.img").c_str());`

Perhaps it is better to do this.

Note that I have not yet bought the RX-888 Mk-2, so I do not know if it works properly.
From what I have heard, the signal-to-noise ratio seems to be quite good.

Anyway, I am grateful that even such a minor model is supported.

Thanks.

JVital2013, 2024-11-22:
Good catches on both! The SDDC is not a module we test often so these have slipped though. Will be fixed shortly

crackerjacques, 2024-11-22:
> Good catches on both! The SDDC is not a module we test often so these have slipped though. Will be fixed shortly

And one more thing, I just compiled it with OrangePi5.

SatDump/plugins/sdr_sources/sddc_sdr_support/lib/Core/fft_mt_r2iq.cpp
SatDump/plugins/sdr_sources/sddc_sdr_support/lib/Core/conv_r2iq.cpp

The part of the code that identifies the CPU in these two codes had to be rewritten.

```
#ifdef _WIN32
    //  Windows, assumed MSVC
    #include <intrin.h>
    #define cpuid(info, x)    __cpuidex(info, x, 0)
    #define DETECT_AVX
#elif defined(__x86_64__)
    //  GCC Intrinsics, x86 only
    #include <cpuid.h>
    #define cpuid(info, x)  __cpuid_count(x, 0, info[0], info[1], info[2], info[3])
    #define DETECT_AVX
#elif defined(__aarch64__)
    // ARM64 specific code
    #define DETECT_NEON
    #include <sys/auxv.h>
    #include <asm/hwcap.h>
    static bool detect_neon()
    {
        return true;
    }
#elif defined(__arm__)
    // ARM32 specific code
    #define DETECT_NEON
    #include <sys/auxv.h>
    #include <asm/hwcap.h>
    static bool detect_neon()
    {
        unsigned long caps = getauxval(AT_HWCAP);
        return (caps & HWCAP_NEON);
    }
#else
    #define NO_SIMD_OPTIM
#endif
```

Thanks.

JVital2013, 2024-11-24:
Do you mind opening this in a PR to make sure we get it right?

Aang23, 2024-12-06:
To be honest, the SDDC stuff is purely left in there for historical reasons. It's not officially supported because there's simply no good API to support that hardware at all. What there is right now barely works, and I've given up on it literally years ago.

Any fix is welcome, but that's the stance on those.

commit d8f5c04b (2024-11-22): Fix #872