SatDump/SatDump #873: SDRPlay RSP1B

AlanB-KG9PN, 2024-11-22:
I just got a SDRplay RSP1B today. SatDump will not recognize it in Ubuntu. I have tried installing the .deb file and building from source. There simply is no option for SDRplay at all. I have installed the API, and all my SDR software other than SatDump sees it and it works fine. Any ideas as to what im doing wrong or what the fix is? (BTW the device is recognized and works fine under Windows.)

JVital2013, 2024-11-23:
Good evening! Can you please post the full log of SatDump opening so we can troubleshoot?

AlanB-KG9PN, 2024-11-23:
I’m fairly new to Ubuntu, how would I find the log?

JVital2013, 2024-11-24:
The log is at ~/.config/satdump (.config is a hidden folder)

AlanB-KG9PN, 2024-11-24:
I appreciate your help. I already figured it out. It was a problem with the API installation. I redid it and all is fine now!

AlanB-KG9PN, 2024-11-24:
Thanks all for the help I figured it out. The first API install didnt go in correctly. It was my error.

closed 2024-11-24 by AlanB-KG9PN