SatDump/SatDump #877: OpenCL Support on Pi4/5 with CLVK Findings and Problems

krakenrf, 2024-11-25:
I noticed that on a Pi 4 SatDump was spending a long time after a pass with all Pi CPUS at 100% creating projections and products.

So I started experimenting with CLVK https://github.com/kpet/clvk, which is a OpenCL implementation built on top of Vulkan. This essentially adds OpenCL support for the Pi 4 and Pi 5 which doesn't have a native OpenCL implementation.

After getting it installed, it's possible to see the V3D device in SatDump settings, as well as an llvmpipe software implementation of OpenCL.

Unfortunately, while SatDump sees it, the OpenCL kernel in SatDump doesn't seem to run on it.

I've been testing by running a projection on a NOAA HRPT AVRR image and I get a bunch of warnings

```
(E) Error warping on GPU : Error building: clvk-yyv0aV/source.cl:34:13: warning: comparing floating point with == or != is unsafe
   34 |   if (shift == 0)
      |       ~~~~~ ^  ~
clvk-yyv0aV/source.cl:37:9: warning: mixing declarations and code is incompatible with standards before C99
   37 |   float x = cos(*lat * DEG_TO_RAD) * cos(*lon * DEG_TO_RAD);
      |         ^
clvk-yyv0aV/source.cl:59:9: warning: mixing declarations and code is incompatible with standards before C99
   59 |   float dist1 = SQ(xr[1] - pxy[0]) + SQ(yr[1] - pxy[1]);
      |         ^
clvk-yyv0aV/source.cl:79:23: warning: implicit conversion from '__private int' to 'float' may lose precision
   79 |   float x_diff = rx - x;
      |                     ~ ^
clvk-yyv0aV/source.cl:80:23: warning: implicit conversion from '__private int' to 'float' may lose precision
   80 |   float y_diff = ry - y;
      |                     ~ ^
clvk-yyv0aV/source.cl:82:29: warning: implicit conversion changes signedness: 'int' to 'size_t' (aka 'unsigned i
```

I'm not familiar with OpenCL kernels, but these just look like C compatibility issues, and I went ahead and fixed them. But in the end it seems that the warnings were just warnings, and not the root cause of the error. I end up with the following output (with no further info given after "error building", and then with it falling back to the CPU.

```
[10:48:20 - 25/11/2024] (I) Update image...
[10:48:20 - 25/11/2024] (I) Done
[10:49:08 - 25/11/2024] (I) Update projection...
[10:49:08 - 25/11/2024] (I) Reprojecting layer 0...
[10:49:08 - 25/11/2024] (D) Filtering timestamps...
[10:49:08 - 25/11/2024] (T) Warping size 20480x10240
[10:49:08 - 25/11/2024] (T) Smart Warp using 20480 2162, full size 20480 10240
[10:49:08 - 25/11/2024] (T) We will split into 1 segments. Median distance is 108.565823 km and total (avg) distance is 3691.237974 km
[10:49:08 - 25/11/2024] (I) Solving TPS equations for 644 GCPs...
[10:49:10 - 25/11/2024] (I) Solved! Took 1.145638
[10:49:11 - 25/11/2024] (D) Using GPU! Double precision requested 0
[10:49:11 - 25/11/2024] (T) First OpenCL context request. Initializing...
[10:49:11 - 25/11/2024] (I) Using platform: clvk
[10:49:11 - 25/11/2024] (I) Using device: V3D 4.2.14
[10:49:14 - 25/11/2024] (E) Error warping on GPU : Error building: 
```

There is also an llvmpipe GPU emulator device added by CLVK. Selecting this device allows the projection calculation to finish entirely on the llvmpipe emulator, but the projected image is missing from the map, so obviously something went wrong.

It would be nice if the Pi4/5 V3D GPU could actually be used, but I'm unsure if it's possible to make the kernels compatible. Does anyone know?

krakenrf, 2024-11-30:
Just wanted to share that I got it working for warp_image_thin_plate_spline_fp32.cl so far at least. The kernel just needs to be modified to eliminate ushort and warp.cpp modified to cast everything to uint32_t and back again to uint16_t after the GPU. 

Currently, the CPU takes 245s to create an 2048x1024 equirectangular projection on my test NOAA HRPT AVRR image, and the GPU 180s. So the Pi 4 GPU is a bit of a potato, but at least it's not hogging four cores during the processing.

I'll upload example code in a few days, I still want to play around a bit and see if it can be optimized.

krakenrf, 2025-01-07:
If anyone's interested, the modified code and kernel to make SatDump projections compatible with CLVK is here https://github.com/krakenrf/SatDump

I made a few more tweaks, and the fastest I can get on a Pi 4 is 80s, which is significantly faster than the 245s the CPU takes. With the GPU overclocked a bit I was able to get it down to 60s. It should be much faster on the Pi 5.

Not sure if there is ever going to be any interest in adding native support to SatDump for CLVK by including modified kernels and caller code, but just in case anyone wants to test it out, here are a few notes on installing CLVK on the Pi 4 as compiling it on the Pi 4 was a bit troublesome I believe due to RAM limits.

First use --shallow when fetching sources otherwise the download is too big 

`./external/clspv/utils/fetch_sources.py --deps llvm --shallow`

Then I had to compile CLVK first on a Linux PC or Pi 5, and copy over clspv64--.bc and clspv--.bc from clvk/build/external/clspv/third_party/llvm/tools/libclc into a folder clvk/libclc on the Pi 4. It seems that these files are too big to successfully compile on the Pi 4, but fortunately, they don't need to be compiled on the same architecture. On the Pi 5 the libclc files compile OK.

Then use 'realpath' to reference the libclc dir during cmake

`cmake ../ -DCMAKE_C_COMPILER=clang -DCMAKE_CXX_COMPILER=clang++ -DCLSPV_EXTERNAL_LIBCLC_DIR=$(realpath ../libclc)`

The use make -j3 to compile, as make -j4 seems to crash.

Then to run SatDump with OpenCL support I used the following. Probably some of these arguments are not required.

```
CLVK_CLSPV_OPTIONS="-O3 -cl-fast-relaxed-math -cl-no-signed-zeros -cl-denorms-are-zero -cl-finite-math-only --cluster-pod-kernel-args" \
CLVK_CLSPV_NATIVE_BUILTINS=sin,cos,log,exp \
CLVK_SPIRV_VALIDATION=0 \
CLVK_MAX_BATCH_SIZE=100 \
CLVK_MAX_CMD_GROUP_SIZE=100 \
CLVK_FORCE_SUBGROUP_SIZE=4 \
CLVK_LOG=4 \
CLVK_LOG_DEST=file:clvk.log \
CLVK_VALIDATION_LAYERS=0 \
CLVK_IGNORE_OUT_OF_ORDER_EXECUTION=0 \
CLVK_CACHE_DIR=/home/dd/clvk_cache \
LD_LIBRARY_PATH=/home/dd/clvk/build/ \
satdump-ui
```

JVital2013, 2025-01-07:
Great research, thanks for the write-up and the code! Hopefully this helps some folks.

It would need some work before we could integrate it into SatDump - the first thing that comes to mind is the 32-bit images immediately increases memory usage by at least 2x, which might be too much of an increase for some use cases/systems when projecting high-res imagery like FCI. Perhaps there's some way we can detect if the OpenCL implementation supports uint16 and enqueue the correct kernel? Also, it would be nice to detect the presence of things like `native_log` and similar, since they are often much faster, but sometimes not reliable. I'm not sure if that's practical, though.

Once we get past that, we would still need to have some how-to based on yours that informs folks how to set up clvk. If we provided .deb files, we could get past the compilation issue, but setting up some environment variables may still be necessary. It would be interesting to find out which of those are really necessary.

I'm just thinking out loud here. I'm sure @Aang23 has other thoughts since he's the main OpenCL expert :-)