SatDump/SatDump #880: SatDump Rotator Auto Reconnect

rtlsdrblog, 2024-11-28:
Every now and then my internet connection or WiFi glitches out momentarily, and SatDump loses the socket connection to my rotator. The rotator reconnects to the WiFi just fine, but SatDump doesn't automatically reconnect the socket. So some passes end up being lost until I manually click on the connect button again.

It would be nice if there was an auto-connect option, where SatDump simply polls the socket every few seconds, and connects to the socket if it can.

Aang23, 2024-12-07:
Are you controlling the rotator over WiFi or the Internet? 

It's perfectly possible to reconnect, but at the same time disconnection issues of the sort on what you would expect to be a local network usually represent something is going pretty wrong, and you wouldn't want a rotator going haywire potentially damaging itself or other hardware if that was the case.

rtlsdrblog, 2024-12-09:
Over WiFi. 

I'm fairly certain it's a specific issue to do with ESP32 and Google WiFi mesh networks, or maybe mesh networks in general. When connected to the Google WiFi mesh, it will disconnect every few hours. When connected to a non-mesh wifi point, it's stable.

But the single non-mesh wifi point does not reach up to my roof.

Having a checkbox 'Auto-reconnect' that maintains the connection after clicking on 'Connect' would help work around this.

Baerchen239, 2025-02-02:
I have also described a problem with the connection to the controller here:
https://github.com/SatDump/SatDump/issues/919
However, SatDump terminates the connection on its own for me.
Perhaps both problems can be solved in one solution?