SatDump/SatDump #881: SatDump crash when trying to decode NOAA19 audio file

Canaribou02, 2024-11-28:
**Description of the issue**
Hello everyone, today i was trying to decode an audio file from NOAA19 which i recorded yesterday, but everytime i try to decode the audio, the progression bar arrive at 100% and suddenly SatDump close without error message and i don't know what to do.
So please help me !

**Hardware (SDR/PC/OS)**
RTL-SDR V3/ Laptop Asus Vivobook x509ja
**Version (Eg, 1.0.0, CI Build 171)**
SatDump version 1.2.1
**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
[
[satdump-20241128T170827.log](https://github.com/user-attachments/files/17950359/satdump-20241128T170827.log)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
This is before i tried to decode the audio
[image]
I can send other screenshots if needed

JVital2013, 2024-11-28:
Hello!

This looks like the crash that was fixed at aedcd32. Please try again using the latest nightly: https://github.com/SatDump/SatDump/releases/tag/nightly

Canaribou02, 2024-11-28:
Thank you for your answer ! But after try with the nightly version i still have the same problem.

JVital2013, 2024-11-28:
Uh oh! Can you share the wav file? You may need to zip it up, or upload it to another site

Canaribou02, 2024-11-28:
Here is the audio file: [https://we.tl/t-kM4YGUsMWX](url)
I already decode it with an another site and the image was pretty good.

JVital2013, 2024-11-28:
Thanks! I just tested your file on the latest nightly, and it didn't crash, but it did hang. It's possible it will crash some systems. I'll get this fixed up

Canaribou02, 2024-11-28:
Ok, thank you for your help !

JVital2013, 2024-11-28:
This should now be fixed in the latest nightly. The error was caused by the code not properly handling when you don't set the start timestamp. It will now throw appropriate warnings, but there should be no negative side effects.

You can now decode the image from your wav, but if you don't mind, I have some feedback:

- Setting the start timestamp is highly recommended. Without that, SatDump doesn't know where the satellite is, and cannot produce the nice MCIR/MSA images with colored land/water.
- Your signal is too noisy to read calibration info, which means most of the composites cannot be generated even with the start timestamp. Antenna design and noise source removal is king here! You are close to a signal good enough to calibrate.
- Also, make sure there is no compression when recording, and that you are sampling enough spectrum - a bandwidth of 40 kHz is usually good. There's something funky going on with the strips on the left - some kind of "shadowing."
 Just why I bring this up!

Overall though, nice initial work!

[image]

closed 2024-11-28 by JVital2013