SatDump/SatDump #885: Critical issue on MacOS

Lolagatorade, 2024-12-03:
**Description of the issue**
failure to load

**Hardware (SDR/PC/OS)**
Macos Big sur aka Macos 11 on a macbook air 13 4gb 2013

**Version (Eg, 1.0.0, CI Build 171)**
Latest version of softwre sunning on macos big sur

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

[image]

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
tried loading the program as admin ect whatever chatgpt reccomended and nothing worked.

JVital2013, 2024-12-03:
Hello! Thanks for reaching out. It looks like you denied SatDump's access to your computer files, which is necessary for its operation. See the instructions here for how to turn on file access for SatDump: https://support.apple.com/guide/mac-help/control-access-to-files-and-folders-on-mac-mchld5a35146/11.0/mac/11.0

Lolagatorade, 2024-12-03:
[image]

Gave the app as many permissions as possible in the security panel but still not loading even after a reboot

Lolagatorade, 2024-12-03:
[image]
[image]
[image]

Seeing folder contents to see if anything can be obtained at a deeper log level

Lolagatorade, 2024-12-03:
[image]

Thank you for pointing out the issue I had no idea a file permissions error was a thing on macOS basically I had to go and add users to the configuration file because I only had certain groups allowed and it wasn't even read and write so I proceeded to just add everything under the table and it worked which is very strange I just know that the permission system on macOS is very weird and I've only experienced something similar on Linux trying to get groups to work with Plex so that I can look at every single file on the system etc. hopefully this can be streamline for other users that might have the same issue on macOS either way thank you very much for the pointers they helped!!

I will leave it open in case this is an issue you might want to address it down the line as simply changing the location of the configuration file or having it be self contained might solve the issue for anybody else since macOS doesn't actually work like a Linux system where for the most part the files are scattered around the system and not self-contained

JVital2013, 2024-12-04:
I'm glad you figured it out - and I'm happy I could help you find the problem!

At this time, I'm not going to make any changes because this permissions issue is specific to your machine. Nobody else has reported this issue, and it tests fine for me on multiple macOS versions. `~/.config` is within the user profile, and the user is supposed to have full read-write access to it. Many applications, including serious ones like Visual Studio for macOS, Inkscape, htop, and many more write to the `~/.config` folder. Applications expecting this to be writable is the norm, so something broke your filesystem permissions at some point in the past.

Admittedly, the preferred location to store user preferences for macOS apps is in `~/Library/Application Support`. Previous versions of SatDump used `~/Library/Application Support`, but we kept running into edge cases depending on how SatDump was built. We moved to the same standard we use on other Unix-like OSs for consistency.

closed 2024-12-04 by JVital2013