SatDump/SatDump #895: SatDump on Android - no signal after version 1.2.1

miclass, 2024-12-22:
Hi all, I am using SatDump on Android but it seems it is not working anymore for me after version 1.2.1. Basically on the waterfall I just have a signal peak on the frequency I tuned and nothing more, as if were not connected to any antenna. I only have dark blue in the waterfall around the tuned frequency, no matter other settings I change (frequency, AGC,...).  If I revert to version 1.2.0 it works again.

My device is OnePlus 10 Pro with Android 14, but I have exactly the same behaviour on a older Lenovo Tablet. I use a Nooelec SDR Smartee, but same happens with rtl-sdr.com. 

I attach two screenshots, first one shows SatDump 1.2.2 with the dark blue waterfall (no signal), the other one shows SatDump 1.2.0 waterfall (here there was no antenna attached, but it's to show the difference in the signal, this version works):

[image]
SatDump 1.2.2 - dark blue waterfall, no signal beside a peak on the tuned frequency

[image]
SatDump 1.2.0 - very difference waterfall...

Thanks for any suggestion!
Please let me know if any other info are needed.
Regards
-miclass IU2RQJ

closed 2025-01-06 by JVital2013

JVital2013, 2025-01-06:
Hello! Thanks for the report.

Long story short, the RTL-SDR has two AGCs:

- An internal LNA AGC often called RTL AGC. This gain works on the RTL2832 chip itself, and is more-or-less its own thing.
- A tuner AGC, which runs on the tuner chip (E3000, R820T2, etc). The tuner AGC automatically overrides the "gain" setting to what the chip thinks is best.

In 1.2.0 and before, the AGC checkbox turned on both of these AGCs. Starting in 1.2.1, I accidentally introduced a bug where it would only set the LNA AGC, not the tuner AGC (oops). On 1.2.1/1.2.2, you can still use it by manually setting the gain instead of relying on the AGC, which is recommended anyway. The tuner AGC can do strange things sometimes.

That being said, this change was a mistake - but rolling it back probably isn't the best. Instead, I just pushed a change to the nightlies that splits out the LNA and tuner AGC settings, so they can be controlled independently. This way, you have full control over the AGC/gain settings.

[image]

Thanks again for the report!

JVital2013, 2025-01-06:
Quick update: it looks like the Android builds are failing at the moment - just keep an eye on https://github.com/SatDump/SatDump/releases/tag/nightly. Once the timestamp updates next to the Android "asset", the new nightly with the fix is ready to download!

JVital2013, 2025-01-07:
As of today this is fixed in the nightly builds

commit b404be22 (2025-01-06): Split out RTL-SDR/TCP LNA/Tuner AGC (fix #895)