SatDump/SatDump #897: EWS-G2 Image reception shifting

diego-biz, 2024-12-25:
Hello, I receive quite well the EWS-G2 satellite (7.6dB SNR) and the processed images results with no corrupted lines or wrong pixels but, as  soon as I start the reception, the preview  has the image shifted left of 1/4 and the processed images where cropped.
I don't know if it is a satellite problem or a SatDump issue: i serach on theweb but the omages from EWS-G2 appears correct.
I'm running the latest stable (1.2.2) with SDRplay RSP2pro.
You can see the Lice Processing thumbnail that has the start of the image at 3/4 of the correct position
[image]
and the result is cropped as follows
[image]

Thanks for your work.

Diego IW3HOI

ik8xld, 2025-05-19:
Same situation here.
after many hours in reception only one or two images are correctly decoded.
The decoding stops randomly and does not conclude the image being acquired, but starts again from the beginning as if it were a new image.
I confirm, during live decoding the image is not centered. 
Hope in a fix

Cpt-Dingus, 2025-08-11:
> [image]
>
> Same situation here.
> after many hours in reception only one or two images are correctly decoded.
> The decoding stops randomly and does not conclude the image being acquired, but starts again [...]

Hi,
This is normal. The progress is only applicable to a **full disk**! It does those at 2:45 UTC and every 3 hours onwards.

The image appearing as 'cut off' is because this is a cropped scan - the imager physically only captures this section of the globe in order to provide it at a higher temporal resolution - 15 minutes as opposed to around 30 for a full disk.

As for why it seems like there are two images, that's just the preview defaulting to the full disk scan resolution - if it used the crop one, a full disk would appear as cropped in it