SatDump/SatDump #899: SatDump infinitely loading

Theovorn, 2024-12-28:
**Description of the issue**
**Prefacing this: I'm writing about this issue on behalf of my father, which uses it.**

SatDump very often freezes on start and never gets past the loading screen.

I tried installing it on my own machine and I have the same issue.

If we insist, and open several instances of SatDump, then somehow one or a few of them finish loading and open correctly.

**Hardware (SDR/PC/OS)**
Win10 (mine) and Win11 (his)
**Version (Eg, 1.0.0, CI Build 171)**
1.2.2 x64 - Correctly installed, with Visual Studio 2019/2022
**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
[satdump-20241228T145221.log](https://github.com/user-attachments/files/18266371/satdump-20241228T145221.log)
^^^ from my PC, when it gets stuck

[satdump-20241228T145309.log](https://github.com/user-attachments/files/18266399/satdump-20241228T145309.log)
^^^ from my PC, when it finishes loading
(making me think that "aaronia_sdr_support.dll" in the beginning of both logs is not the blocking error...)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
[image]
^^^top left is the first one I opened, the others were open afterwards, and all of a sudden, the first instance completed the loading

**Im going to keep the program installed on my PC so that if you'd suggest solutions or else I can try and verify them.**

**Thank you in advance and please go easy on me if its an obvious solution, I'm no expert about these radio-amateur stuff at all!**

JVital2013, 2025-01-04:
Try deleting `C:\Program Files\SatDump\lib\satdump\plugins\remote_sdr_support.dll` and see if that solves your issue. There seem to be some proxy/vpn software that causes problem with this plugin.

Theovorn, 2025-01-05:
I did it and it works now.
Thank you very much!

closed 2025-01-05 by Theovorn