SatDump/SatDump #900: Incorrect path concatenation when launching v1.2.2 on NixOS

maeve-oake, 2024-12-28:
**Description of the issue**
After packaging and building satdump for NixOS, packages seem to get loaded from an incorrectly formed path which causes a crash:
```
[14:24:27 - 28/12/2024] (I) Loading plugins from /nix/store/zbvsfqfp7hcm3cqx9lhbxqlrvp94xhkn-satdump//nix/store/zbvsfqfp7hcm3cqx9lhbxqlrvp94xhkn-satdump/lib/satdump/plugins
```

**Hardware (SDR/PC/OS)**
Framework 13 - AMD Ryzen 7840U - NixOS

**Version (Eg, 1.0.0, CI Build 171)**
Latest build commit (v1.2.2)

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
```
maeve@replika ~ [1]> satdump-ui
[14:24:26 - 28/12/2024] (E) Usage : satdump-ui [downlink] [input_level] [input_file] [output_file_or_directory] [additional options as required]
[14:24:26 - 28/12/2024] (E) Extra options (examples. Any parameter used in modules can be used here) :
[14:24:26 - 28/12/2024] (E)  --samplerate [baseband_samplerate] --baseband_format [cf32/cs32/cs16/cs8/wav16] --dc_block --iq_swap
[14:24:26 - 28/12/2024] (D) Starting with OpenGL 4.6 (Compatibility Profile) Mesa 24.2.6
[14:24:26 - 28/12/2024] (D) Max texture size: 16384
[14:24:26 - 28/12/2024] (E) Glfw Error 65548: Wayland: The platform does not support setting the window icon
[14:24:26 - 28/12/2024] (I)    _____       __  ____                      
[14:24:26 - 28/12/2024] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[14:24:26 - 28/12/2024] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[14:24:26 - 28/12/2024] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[14:24:26 - 28/12/2024] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[14:24:27 - 28/12/2024] (I)                                     /_/      
[14:24:27 - 28/12/2024] (I) Starting SatDump v1.2.2
[14:24:27 - 28/12/2024] (I) 
[14:24:27 - 28/12/2024] (I) Loading config /nix/store/zbvsfqfp7hcm3cqx9lhbxqlrvp94xhkn-satdump/share/satdump/satdump_cfg.json
[14:24:27 - 28/12/2024] (W) No user configuration found! Keeping defaults.
[14:24:27 - 28/12/2024] (I) Loading plugins from /nix/store/zbvsfqfp7hcm3cqx9lhbxqlrvp94xhkn-satdump//nix/store/zbvsfqfp7hcm3cqx9lhbxqlrvp94xhkn-satdump/lib/satdump/plugins
terminate called after throwing an instance of 'std::filesystem::__cxx11::filesystem_error'
  what():  filesystem error: recursive directory iterator cannot open directory: No such file or directory [/nix/store/zbvsfqfp7hcm3cqx9lhbxqlrvp94xhkn-satdump//nix/store/zbvsfqfp7hcm3cqx9lhbxqlrvp94xhkn-satdump/lib/satdump/plugins]
fish: Job 1, 'satdump-ui' terminated by signal SIGABRT (Abort)
```

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
Nix derivation:
```
{ pkgs, stdenv, lib, fetchFromGitHub }:

stdenv.mkDerivation rec {
  name = "satdump";
  version = "1.2.2";

  src = fetchFromGitHub {
    owner = "SatDump";
    repo = "SatDump";
    rev = "1.2.2";
    sha256 = "sha256-+Sne+NMwnIAs3ff64fBHAIE4/iDExIC64sXtO0LJwI0=";
  };

  nativeBuildInputs = with pkgs; [ cmake pkg-config ];
  buildInputs = with pkgs; [
    # required deps
    fftwFloat
    libpng
    libtiff
    jemalloc
    volk
    (nng.overrideAttrs (old: {
      cmakeFlags = old.cmakeFlags ++ [ "-DBUILD_SHARED_LIBS=ON" ];
    }))
    rtl-sdr-librtlsdr
    hackrf
    airspy
    airspyhf
    glfw
    zenity
    zstd
    curl

    opencl-headers
    ocl-icd
    rocmPackages.clr.icd

    # optional hw support
    libad9361
    libiio
    # sdrplay
    # limesuite
    # libbladeRF
    # uhd
    # hdf5
  ];

  cmakeFlags = [
    "-DCMAKE_BUILD_TYPE=Release"
  ];
}
```

theverygaming, 2024-12-29:
I almost created a dupe issue^^ - good that i checked. I am preparing to create a nixpkgs PR for SatDump right now and ran into this.

The issue is this line here:
https://github.com/SatDump/SatDump/blob/91c411daf010e62a66c437805cd0f43b5b2158aa/src-core/CMakeLists.txt#L37

`CMAKE_INSTALL_LIBDIR` is an absolute path for reasons i have not figured out (and will probably not investigate further) in the nix CMake build environment, a bunch of other nix packages seem to work around it with patches to use `CMAKE_INSTALL_FULL_LIBDIR` instead of `${CMAKE_INSTALL_PREFIX}/${CMAKE_INSTALL_LIBDIR}`

theverygaming, 2024-12-31:
There is some further reference on this - see https://github.com/jtojnar/cmake-snips?tab=readme-ov-file#assuming-cmake_install_dir-is-relative-path

@Aang23 i will solve this with a patch for now but can this be fixed in the next release-ish? Simply using `CMAKE_INSTALL_FULL_LIBDIR` instead seems to work fine on Debian too.

JVital2013, 2025-01-04:
Thanks for the notes @theverygaming! I tested CMAKE_INSTALL_FULL_LIBDIR and it seems to work correctly across all operating systems where it gets used, so I changed the cmake file.

Thanks again!

commit e8f697d1 (2025-01-04): Do not assume relative CMAKE_INSTALL_LIBDIR (Fix #900)
closed 2025-01-04 by JVital2013