SatDump/SatDump #905: Compilation problem on Intel Mac with Sequoia 15.2

gregtravis, 2025-01-04:
I'm getting the following when trying to compile SatDump on an Intel 27" iMac running Sequoia 15.2

"/Users/greg/Desktop/SatDump/src-core/./libs/sol2/sol.hpp:19875:31: error: address of overloaded function 'call' does not match required type 'int (lua_State *)'"

I've made sure all of my home-brew dependencies are up to date as well as Xcode and its command-line tools (Xcode v. 16.2)

I'm using the steps outlined for building under MacOS, namely:

1. git clone https://github.com/SatDump/SatDump.git && cd SatDump
    --- Works
2.  ./macOS/Configure-vcpkg.sh
    --- Works
3.  mkdir build && cd build
   ---- Works
4.  cmake -DCMAKE_TOOLCHAIN_FILE=../vcpkg/scripts/buildsystems/vcpkg.cmake -DCMAKE_BUILD_TYPE=Release -DVCPKG_TARGET_TRIPLET=osx-satdump ..
    ---- Works
5.   make -j$(sysctl -n hw.logicalcpu)
    ---- Fails with the above message

gregtravis, 2025-01-04:
FWIW I followed this suggestion:

"I have this issue too. For what it's worth, replacing upvalue_this_member_variable::call()'s noexcept(std::is_nothrow_copy_assignable_v<T>) specification by either noexcept(true), noexcept(false) or even removing it entirely seems to "fix" it for some reason. I don't really understand that behavior right now, but maybe that can help a bit."

And it seems to have fixed the issue.  And I echo the "I don't really understand that behavior" sentiment

JVital2013, 2025-01-04:
Thanks for testing, and I'm glad you got it to build!

It's a bug in llvm - they fixed it, but the fix has not made it's way into the latest XCode as far as I can tell. It should soon though. I'll have to think about if we should patch the library to accommodate the temporarily broken compiler, or just wait for Apple to fix it. They're usually quite good about that.

gregtravis, 2025-01-04:
It does build and run on my iMac now.  However the GUI is all messed up and the waterfall doesn't work at all.  I may see if it's a hardware issue by building and running on my Apple Silicon MacBook (also running Sequoia 15.2) just to rule out a hardware issue on the iMac

In the meantime, I'll stick with the pre-compiled releases for production ;-)
[image]

JVital2013, 2025-01-11:
Oh yikes - that OpenGL code did _not_ build correctly. I'm curious if there's any interesting errors in Console.app when launching your self-built app? Or, if you launch ./satdump-ui in the terminal?

Either way, it looks like we need to research changes with XCode 16 since I think that may be the culprit.

EricFROL, 2025-02-08:
same issue here. any workaround?

JVital2013, 2025-02-08:
Not yet - I believe it works if you can roll back to XCode 15.4, but that's untested.

I just today got access to a newer mac. I hope to troubleshoot this issue quite soon!

JVital2013, 2025-02-09:
OK, I've done a bit of testing on the latest macOS/Xcode. Which issue are you getting, @EricFROL: the "red screen" error, or the sol.hpp build error? The sol.hpp build error can be worked around in one of two ways:

1. Roll back to Xcode 15.4
2. Apply one of the patches linked in my first reply to `src-core/libs/sol2/sol2.hpp`

The sol.hpp issue is really a bug in Xcode that Apple is aware of, which is why I don't want to apply those patches to SatDump (unless this drags on too long...) But, at least there is a workaround for that.

On the other hand, the "red screen" shown by @gregtravis I cannot reproduce. I am on a newer version of macOS than you reported, though - I'm on macOS 15.3, but still with Xcode 16.2. Maybe that bug will go away with an OS update?

[image]