SatDump/SatDump #907: Linux - SatDump - RSP1a

Michal259, 2025-01-08:
Hello,
Three days ago I installed SatDump (satdump_1.2.2_ubuntu_22.04_amd64.deb) on my computer. 
It looks good and will be useful for my hobby. Thanks to all the programmers.
I am using Linux Mint 21.3 based on Ubuntu 22.04 jammy.
I have SDR RSP1A from SDRplay. 
When I run satdump-ui, in Recorder - Device I don't have the option to select my SDR receiver.
So far I have only been able to get the connection working via RTL-TCP and rsp_tcp.
How is it possible to make it so that I have my SDR receiver directly in the SatDump menu?
Thank you for your help.
mfa

JVital2013, 2025-01-09:
At this time, you must build Satdump yourself using the build instructions in the main readme for SDRPlay support on Ubuntu.

Make sure you uninstall the deb version first, and you have the SDRPlay API installed before building the software.

Michal259, 2025-01-09:
Thank you for your reply.
I'm going to install the missing dependencies and I'll try the build.
Then I will write if I was successful.
:)

Michal259, 2025-01-09:
It succeeded and the build went through, although there were quite a lot of warnings.
I can already see my SDR receiver RSP1A in the devices menu.
Now I'll wait about half an hour for the NOAA18 flyover and I'll monitor the signal on 137.1 MHz.
Thanks for the advice.
mfa

JVital2013, 2025-01-11:
Yeah, there are a number of warnings on build right now. They can be ignored for the most part - some of them relate to newer libs we've embedded that aren't completely tamed, and others are bugs in some of the more niche features or stuff that isn't fully used at the moment. We'll clean our house eventually :-)

Any luck with the signal?

Michal259, 2025-01-11:
Receiving signals from NOAA satellites 15, 18, 19 on 137 MHz works very
nicely.
I still have to make an antenna for the 1700 MHz band :)

mfa