SatDump/SatDump #908: Process Multiple Pipelines (Frequencies) from a Single Device (SDR)

reishe00, 2025-01-11:
**What would you want to be added to satdump?**
Currently in the live "Recorder" tab, there an option to select one **device**, one **frequency**, and one **pipeline** per session.
This can be a limitation when simultaneously decoding multiple narrow bandwidth Inmarsat pipelines (600bps, 1200bps, 10,500bps).
Currently, we need multiple physical SDRs, and multiple instances of SatDump, regardless of the SDR bandwidth available.

As below, all the frequencies are within a bandwidth of 1.25Mhz. In theory, it should be possible to decode both the 0.6K and 10.5K channels simultaneously using one SDR.

[image]

**Describe the feature you'd like to have**
Ideally, the feature would allow decoding of multiple pipelines/frequencies within the bandwidth of a single selected device (SDR).

**Describe alternatives you've considered**
The only alternative I have explored is using multiple sessions of Satdump each with its own physical SDR, but this could be costly depending on the number of signals to decode.

JVital2013, 2025-01-11:
Hello! Thanks for the feedback.

This is a planned feature. It's partially implemented, but it only works if you're using the scheduler - and that's mostly for LEO/MEO satellites, and does not work well for GEOs like Inmarsat. We do plan to swing back to it though!

Aang23, 2025-01-12:
Actually, if using the CLI VFOs can be used to do exactly this.