SatDump/SatDump #914: SatDump creates an empty `Unknown` folder when running geostationary pipelines

Cpt-Dingus, 2025-01-18:
**Description of the issue**
SatDump creates an `Unknown` folder inside the `IMAGES` directory when running geostationary pipelines, tested with Elektro xRIT, GK-2A xRIT, and GOES HRIT.
**Hardware (SDR/PC/OS)**

**Version (Eg, 1.0.0, CI Build 171)**
Happens as early as SatDump 1.2.2, still happens on latest commit.

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
N/A

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

[image]

alsldlflglhljlk, 2025-03-23:
This is, in fact, not a bug. For GOES HRIT specifically, the HRIT downlink does not send data directly from the satellite's instuments. Instead, GOES sends the raw data down to a NOAA ground station, then the ground station sends processed data back to GOES. The HRIT downlink actually sends pre-processed data from NOAA. As such, NOAA can send anything they want up to GOES, like other text files and data from other satellites. On the (somewhat rare) occasion that NOAA sends something that SatDump does not recognize, it puts it in the said `Unknown` folder. I assume this is why it is created for the other broadcasts as well.

Cpt-Dingus, 2025-03-23:
@alsldlflglhljlk This is not exclusive to GOES-R HRIT. This would also mean that there would be data within the folder, which there is not. It also happens without syncing any signal at all.

alsldlflglhljlk, 2025-03-23:
So the issue is that SatDump creates the `Unknown` folder when there isn't actually any unknown data?

Cpt-Dingus, 2025-03-23:
Correct. It's created erroneously.