SatDump/SatDump #917: Tuner AGC disables LNA Gain bargraph

f6bvp, 2025-01-20:
Using SatDump remote server and local satdump-ui

**Description of the issue**
When selecting Tuner AGC the LNA Gain bargraph is disabled

**Version (Eg, 1.0.0, CI Build 171)**
SatDump v1.2.3-5e01df112

Clicking on LNA AGC : 
11:27:30 - 20/01/2025] (D) Set RTL-SDR Tuner gain mode to 1
[11:27:30 - 20/01/2025] (D) Set RTL-SDR Gain to 49.6

Clicking on Tuner AGC : 
11:29:04 - 20/01/2025] (D) Set RTL-SDR AGC to 0
[11:29:04 - 20/01/2025] (D) Set RTL-SDR Tuner gain mode to 0

and LNA Gain bargraph cannot be changed any more.

Possibly inversion between  LNA and Tuner AGC  buttons ?

Regards,

Bernard

f6bvp, 2025-01-27:
Additional remark : 
When LNA AGC is selected, LNA Gain bargraph can still be moved up or down. One would expect it to be blocked.

This seems to confirms that LNA AGC and Tuner AGC labels are inverted.

Bernard

JVital2013, 2025-01-30:
Hello,

Good catch! However, the LNA AGC and Tuner AGC labels aren't inverted - the problem is the wrong label is on the gain slider, which should be labeled "tuner gain".

RTL-SDRs do not have LNA gain control, only tuner. Technically, E4000 tuners also have "Tuner IF" gain control, but so far that has not been implemented.

This will be fixed in the next nightly. Thanks!

closed 2025-01-30 by JVital2013