SatDump/SatDump #922: volk not installed by ".\windows\Configure-vcpkg.ps1"

TurboBG, 2025-01-30:
Hello and thank you for this project,

**Description of the issue**
After using ".\windows\Configure-vcpkg.ps1", I tried to build the project with Visual Studio 2022 on Windows 11. I had errors mentioning "volk/volk.h" not found.  It appeared that no volk folder has been created in the project folder.

I tried to manually copy volk and vulkan folders in ".\SatDump\src-core" and now have "fatal error LNK1104: cannot open file 'volk.lib" on visual studio and  "fatal error LNK1184: impossible to open file 'volk.lib"

I created a new folder and tried ".\windows\Configure-vcpkg.ps1" again, but same result.

Did I miss a requirement or a step?

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
[Vs_logs.txt](https://github.com/user-attachments/files/18600143/Vs_logs.txt)
[Vs_logs2.txt](https://github.com/user-attachments/files/18600166/Vs_logs2.txt)
[Powershell_logs.txt](https://github.com/user-attachments/files/18600208/Powershell_logs.txt)

JVital2013, 2025-01-30:
Hello!

The full output from `.\windows\Configure-vcpkg.ps1` would be needed to troubleshoot this, but usually when volk ends up not being installed, it's because mako was not installed with PyPI/pip before running Configure-vcpkg.ps1.

I'd recommend deleting your vcpkg folder in the SatDump directory, run "pip3 install mako", then re-run `.\windows\Configure-vcpkg.ps1`

TurboBG, 2025-01-31:
Thank you for your answer ! 

I just checked and Mako is already installed.
I will try `.\windows\Configure-vcpkg.ps1` again when I find an alternative to `Start-transcript` to log the output (it does not log everything given by Configure-vcpkg.ps1)

TurboBG, 2025-02-04:
Sometimes, the best way to solve a problem is just to dodge it ! After 5 days of debugging, I switched to a computer running Ubuntu and everything went extremly smoothly  !

@JVital2013 Thank you for your help

closed 2025-06-25 by JVital2013