SatDump/SatDump #928: Settings not being saved up

EricFROL, 2025-02-08:
**Description of the issue**
Latitude and longitude arent being saved up when closing and opening Satdump. No error on console log, in fact, I get "Savig user config at...".
Settings changed not being saved.

**Hardware (SDR/PC/OS)**
MacOS 15.2 (24C101)

[Release 1.2.2](https://github.com/SatDump/SatDump/releases/tag/1.2.2)
Precompiled DMG.
Also tried building, getting this https://github.com/SatDump/SatDump/issues/905

JVital2013, 2025-02-08:
I'm suspicious it's a permissions issue - is anything being saved at ~/.config/satdump? If you're unfamiliar with it, ~/.config is a hidden folder so using Terminal is your best bet to get there

JVital2013, 2025-02-08:
For what it's worth, on a machine with a fresh install of macOS Sequoia, I ran into an issue with the permissions on my ~/.config folder. Another program I installed first, and ran with sudo, set its permissions to root only. I was able to fix it by running this command:

```sudo chown -R <username>:staff ~/.config```

EricFROL, 2025-02-24:
yea its working now. thank you and sorry for late response.

closed 2025-02-24 by EricFROL