SatDump/SatDump #932: See Black Lines (missing data)

njrusty, 2025-02-22:
When I run the commend:
goesrecv -v -i 1 -c ~/goesrecv.conf 

My vit numbers are averaging 475 (which is great) and it is showing 0 drops, but all my full images (GOES 16) is showing blackouts of large portions of Earth.

njrusty, 2025-02-23:
I reinstalled SatDump and now I am getting the following error message

Linux GOES 6.6.47+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.47-1+rpt1 (2024-09-02) aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Feb 23 12:46:31 2025
pi@GOES:~ $ satdump live goes_hrit GOES_16 --source rtlsdr --samplerate 2.4e6 --frequency 1694.1e6 --gain 49 --http_server 192.168.1.8:8080 --fill_missing
[17:50:34 - 23/02/2025] (I)    _____       __  ____
[17:50:34 - 23/02/2025] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____
[17:50:34 - 23/02/2025] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[17:50:34 - 23/02/2025] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[17:50:34 - 23/02/2025] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/
[17:50:34 - 23/02/2025] (I)                                     /_/
[17:50:34 - 23/02/2025] (I) Starting SatDump v1.2.3
[17:50:34 - 23/02/2025] (I)
[17:50:34 - 23/02/2025] (I) Loading config /usr/share/satdump/satdump_cfg.json
[17:50:34 - 23/02/2025] (I) Loading user settings /home/pi/.config/satdump/settings.json
[17:50:34 - 23/02/2025] (I) Loading plugins from /usr/local/lib/satdump/plugins
terminate called after throwing an instance of 'std::filesystem::__cxx11::filesystem_error'
  what():  filesystem error: recursive directory iterator cannot open directory: No such file or directory [/usr/local/lib/satdump/plugins]
Aborted
pi@GOES:~ $

alsldlflglhljlk, 2025-03-16:
This is not a problem or question regarding `SatDump`, rather, it concerns `goestools`. Please post this at a more relevant forum.

alsldlflglhljlk, 2025-03-23:
Your Viterbi rate also isn't that great. According to [usradioguy.com](https://usradioguy.com/viterbi-r-s-errors-packets):

> A rate of ~1000 may give you sporadic packets. A rate of ~400 can give you the complete stream of packets without any drops, but is very sensitive to interference. A rate of ~100 or lower is good and can give you a packet stream without drops for hours on end. The lower the Viterbi error rate, the better your signal.

Your rate of ~475 is likely too high to get you the "complete stream of packets" you desire.

Aang23, 2025-06-22:
I can confirm, this is not the repo for `goestools`. SatDump can also decode goes however.

closed 2025-06-22 by Aang23