SatDump/SatDump #936: I cannot start my satdump:(

qkqwork, 2025-03-09:
When I tried to start my satdump, it got stuck at the startup page and displayed "0 TLE LOADED"，and then there was no response:(

Hardware:Windows11,RTL-SDR
Version：1.2.2（Latest）
Log:
```
…
[15:19:47 - 09/03/2025] (D)  - wsf_m_db
[15:19:47 - 09/03/2025] (D)  - xmm_newton_s_link
[15:19:47 - 09/03/2025] (D)  - youthsat_dump
[15:19:47 - 09/03/2025] (D)  - yunhai_ahrpt
[15:19:47 - 09/03/2025] (I) Found OpenCL Devices (1) :
[15:19:47 - 09/03/2025] (D)  - NVIDIA GeForce RTX 3060 Laptop GPU
[15:19:47 - 09/03/2025] (I) Loading TLEs from ./config/satdump_tles.txt
[15:19:47 - 09/03/2025] (I) 0 TLEs loaded!

```

qkqwork, 2025-03-10:
[satdump-20250309T231947.log](https://github.com/user-attachments/files/19155441/satdump-20250309T231947.log)

qkqwork, 2025-03-10:
[image]

nikeedev, 2025-03-23:
Hi! 
Please try opening SatDump through a terminal using CMD or PowerShell. 
Ususally the program reports any issues in there. Just run `satdump-ui`, likely from the same folder as executable.

nikeedev, 2025-04-01:
> * This file is part of `remote_sdr_support`. I then deleted the corresponding DLL `remote_sdr_support.dll` from the `plugins` directory.
> * SatDump is now starting again.

Huh…sounds like some binaries got left behind in the repo…

A script is possibly not allowing the binary to be overwritten…