SatDump/SatDump #938: Plugin Loading Issue on NixOs

prinzhoefte, 2025-03-14:
**Description of the issue**
When installing satdump via this nix flake it only brings up this message when starting and instantly closes:
`
[20:22:04 - 14/03/2025] (I) Loading plugins from /nix/store/hw4ws0akzf3raiknb3lnx3q2d20d4xi8-satdump-nightly//nix/store/hw4ws0akzf3raiknb3lnx3q2d20d4xi8-satdump-nightly/lib/satdump/plugins
terminate called after throwing an instance of 'std::filesystem::__cxx11::filesystem_error'
  what():  filesystem error: recursive directory iterator cannot open directory: No such file or directory [/nix/store/hw4ws0akzf3raiknb3lnx3q2d20d4xi8-satdump-nightly//nix/store/hw4ws0akzf3raiknb3lnx3q2d20d4xi8-satdump-nightly/lib/satdump/plugins]
fish: Job 1, 'satdump-ui' terminated by signal SIGABRT (Abort)
`

NixFile:
`
{ config, pkgs, ... }:

let
  satdump_version = "nightly";
in
{
  environment.systemPackages =
    let
      satdump = with pkgs;
        stdenv.mkDerivation rec {
          pname = "satdump";
          version = satdump_version;

          src = fetchgit {
            url = "https://github.com/SatDump/SatDump.git";
            rev = satdump_version;
            sha256 = "sha256-PwtdkO3FtcZ++7z9LeVkDt8lg7Pn3b08G8C89IAUuMA=";
          };

          nativeBuildInputs = [ cmake pkg-config ];
          buildInputs = [
            # required deps
            fftwFloat
            libpng
            curl
            libtiff
            jemalloc
            volk
            (nng.overrideAttrs (old: {
              cmakeFlags = old.cmakeFlags ++ [ "-DBUILD_SHARED_LIBS=ON" ];
            }))
            rtl-sdr-librtlsdr
            hackrf
            airspy
            airspyhf
            glfw
            zenity
            zstd

            # optional hw support
            libad9361
            libiio
          ];

          cmakeFlags = [
            "-DCMAKE_BUILD_TYPE=Release"
          ];
        };
    in
    [ satdump ];
}

`

Only happens in version 1.2.1 and higher. 1.2.0 is just fine.

prinzhoefte, 2025-05-09:
Fixed in the nightly version by now.

closed 2025-05-09 by prinzhoefte

theverygaming, 2025-05-25:
Just FYI - this is essentially a dupe of #900 which has been fixed in SatDump Nightly

SatDump 1.2.2 is now packaged and avaiable in nixpkgs unstable, where a patch was applied to fix this issue