SatDump/SatDump #939: Compilation error

f1ixx, 2025-03-18:
Hello everyone

First of all, thank you very much for this excellent open-source software.
I'm using it to decode Noaa-APT and try other image formats (frequency, mode, etc.).

I upgraded from Fedora 41 to Fedora 42 beta. The problem is with the plugins.
I'm getting errors, but my skills are too basic for this kind of problem.

I'm make this :
 <pre>cd Satdump
mkdir build && cd build
cmake ../
make
</pre>
Have some warrning but ok and stop with :
<pre>
[ 66%] Built target stereo_support
[ 67%] Building CXX object plugins/inmarsat_support/CMakeFiles/inmarsat_support.dir/aero/acars_parser.cpp.o
[ 67%] Building CXX object plugins/inmarsat_support/CMakeFiles/inmarsat_support.dir/aero/decode_utils.cpp.o
[ 67%] Building C object plugins/inmarsat_support/CMakeFiles/inmarsat_support.dir/aero/libacars/acars.c.o
[ 67%] Building C object plugins/inmarsat_support/CMakeFiles/inmarsat_support.dir/aero/libacars/adsc.c.o
/home/[myPath]/SatDump/plugins/inmarsat_support/aero/libacars/adsc.c: Dans la fonction « la_adsc_contract_request_destroy »:
/home/[myPath]/SatDump/plugins/inmarsat_support/aero/libacars/adsc.c:1365:52: erreur: passage de l'argument 2 de « la_list_free_full » depuis un type pointeur incompatible [-Wincompatible-pointer-types]
 1365 |                 la_list_free_full(r->req_tag_list, la_adsc_tag_destroy);
      |                                                    ^~~~~~~~~~~~~~~~~~~
      |                                                    |
      |                                                    void (*)(void *)
Dans le fichier inclus depuis /home/[myPath]/SatDump/plugins/inmarsat_support/aero/libacars/adsc.c:15:
/home/[myPath]/SatDump/plugins/inmarsat_support/./aero/libacars/list.h:27:43: note: expected « void (*)(void) » but argument is of type « void (*)(void *) »
   27 | void la_list_free_full(la_list *l, void (*node_free)());
      |                                    ~~~~~~~^~~~~~~~~~~~
/home/[myPath]/SatDump/plugins/inmarsat_support/aero/libacars/adsc.c:1344:13: note: « la_adsc_tag_destroy » declared here
 1344 | static void la_adsc_tag_destroy(void *tag) {
      |             ^~~~~~~~~~~~~~~~~~~
/home/[myPath]/SatDump/plugins/inmarsat_support/aero/libacars/adsc.c: Dans la fonction « la_adsc_format_text »:
/home/[myPath]/SatDump/plugins/inmarsat_support/aero/libacars/adsc.c:1784:40: erreur: passage de l'argument 2 de « la_list_foreach » depuis un type pointeur incompatible [-Wincompatible-pointer-types]
 1784 |         la_list_foreach(msg->tag_list, la_adsc_tag_output_text, &ctx);
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~
      |                                        |
      |                                        void (*)(const void *, void *)
/home/[myPath]/SatDump/plugins/inmarsat_support/./aero/libacars/list.h:25:41: note: expected « void (*)(void) » but argument is of type « void (*)(const void *, void *) »
   25 | void la_list_foreach(la_list *l, void (*cb)(), void *ctx);
      |                                  ~~~~~~~^~~~~
/home/[myPath]/SatDump/plugins/inmarsat_support/aero/libacars/adsc.c:1734:13: note: « la_adsc_tag_output_text » declared here
 1734 | static void la_adsc_tag_output_text(void const *p, void *ctx) {
      |             ^~~~~~~~~~~~~~~~~~~~~~~
/home/[myPath]/SatDump/plugins/inmarsat_support/aero/libacars/adsc.c: Dans la fonction « la_adsc_format_json »:
/home/[myPath]/SatDump/plugins/inmarsat_support/aero/libacars/adsc.c:1803:40: erreur: passage de l'argument 2 de « la_list_foreach » depuis un type pointeur incompatible [-Wincompatible-pointer-types]
 1803 |         la_list_foreach(msg->tag_list, la_adsc_tag_output_json, &ctx);
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~
      |                                        |
      |                                        void (*)(const void *, void *)
/home/[myPath]/SatDump/plugins/inmarsat_support/./aero/libacars/list.h:25:41: note: expected « void (*)(void) » but argument is of type « void (*)(const void *, void *) »
   25 | void la_list_foreach(la_list *l, void (*cb)(), void *ctx);
      |                                  ~~~~~~~^~~~~
/home/[myPath]/SatDump/plugins/inmarsat_support/aero/libacars/adsc.c:1749:13: note: « la_adsc_tag_output_json » declared here
 1749 | static void la_adsc_tag_output_json(void const *p, void *ctx) {
      |             ^~~~~~~~~~~~~~~~~~~~~~~
/home/[myPath]/SatDump/plugins/inmarsat_support/aero/libacars/adsc.c: Dans la fonction « la_adsc_destroy »:
/home/[myPath]/SatDump/plugins/inmarsat_support/aero/libacars/adsc.c:1813:42: erreur: passage de l'argument 2 de « la_list_free_full » depuis un type pointeur incompatible [-Wincompatible-pointer-types]
 1813 |         la_list_free_full(msg->tag_list, la_adsc_tag_destroy);
      |                                          ^~~~~~~~~~~~~~~~~~~
      |                                          |
      |                                          void (*)(void *)
/home/[myPath]/SatDump/plugins/inmarsat_support/./aero/libacars/list.h:27:43: note: expected « void (*)(void) » but argument is of type « void (*)(void *) »
   27 | void la_list_free_full(la_list *l, void (*node_free)());
      |                                    ~~~~~~~^~~~~~~~~~~~
/home/[myPath]/SatDump/plugins/inmarsat_support/aero/libacars/adsc.c:1344:13: note: « la_adsc_tag_destroy » declared here
 1344 | static void la_adsc_tag_destroy(void *tag) {
      |             ^~~~~~~~~~~~~~~~~~~
make[2]: *** [plugins/inmarsat_support/CMakeFiles/inmarsat_support.dir/build.make:121: plugins/inmarsat_support/CMakeFiles/inmarsat_support.dir/aero/libacars/adsc.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2754: plugins/inmarsat_support/CMakeFiles/inmarsat_support.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
</pre>

I'm use : gcc 15.0 - Fedora 15
What i'm lost? Package? lib?

Many Thank's

Boris

nikeedev, 2025-03-23:
Hi!
Did you delete everything inside the `build` folder before building the program after the distro update?
If not do so, because using the same cmake files can break the building process as some packages perhaps have been updated but the cmake files may think that you still use the old ones.

According to the docs, you should run this command: `cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr ..`, it has perhaps a few arguments which you did not pass, could be that it will solve the problem?

alsldlflglhljlk, 2025-03-23:
According to the [release page for GCC 15](https://gcc.gnu.org/gcc-15/changes.html):

> Note: GCC 15 has not been released yet, so this document is a work-in-progress.

Please use a stable, released version (14.2.1).

f1ixx, 2025-03-24:
Hello all

Thanks for you back issue.
I erase zvery time the old compil version and build from zero.
I try to roll back gcc.

Thanks
Good day

Boris

closed 2025-03-24 by f1ixx