SatDump/SatDump #940: AppImage target in SatDump releases

alsldlflglhljlk, 2025-03-23:
**What would you want to be added to satdump?**

I wish there was an AppImage download for x86_64 and arm64 systems in the SatDump releases.

**Describe the feature you'd like to have**

It's not really a feature added to SatDump itself, but providing AppImage downloads would make using SatDump considerably easier, especially on systems that aren't Debian or Android based (Arch, Fedora, Manjaro, etc.) AppImages would use more disk space and would be slower to start, but would mostly eliminate the need for users to compile SatDump themselves to use it on an unsupported system.

I imagine the final image would have a `--gui` option to enable using the GUI, but would otherwise use the cli version. Or there could additionally be a CLI only version for users without a GUI or for those wanting to save disk space.

**Describe alternatives you've considered**

#771 mentions adding more distro-specific targets, but this would unnecessarily increase the already large number of build targets that SatDump has.

**Additional context**
Here you can for provide more information about what you'd like to see as the end result. Feel free to also post images if you have ideas regarding the UI

N / A

nikeedev, 2025-03-25:
Satdump uses a lot of dependencies, and the AppImages won't be just large, they will be enourmous. A problem is that a lot of unnecessary features have to be enabled by default (because one cannot customize already built AppImage). Another thing, is that AppImages do not seem to have an easy process of making them...

Aang23, 2025-04-06:
An AppImage is now available!

closed 2025-04-06 by Aang23

nikeedev, 2025-04-06:
> An AppImage is now available!

Hi! Where could one find it? I saw in the commits that you updated the CI file, but, will the AppImage be made only next time you will release a new “Release”, or is it available outside of that?

alsldlflglhljlk, 2025-04-11:
@nikeedev Check the latest nightly build :)

nikeedev, 2025-04-11:
> [@nikeedev](https://github.com/nikeedev) Check the latest nightly build :)

Have you seen it? There is not AppImage there

alsldlflglhljlk, 2025-04-13:
I'm not quite sure what you're talking about. On my end, I can clearly see an AppImage download...

[image]

But, I wouldn't recommend using it quite yet, as its OpenGL bindings seem to be broken as of now.

```
[03:32:51 - 13/04/2025] (E) Usage : satdump-ui [downlink] [input_level] [input_file] [output_file_or_directory] [additional options as required]
[03:32:51 - 13/04/2025] (E) Extra options (examples. Any parameter used in modules can be used here) :
[03:32:51 - 13/04/2025] (E)  --samplerate [baseband_samplerate] --baseband_format [cf32/cs32/cs16/cs8/wav16] --dc_block --iq_swap
[03:32:51 - 13/04/2025] (E) Glfw Error 65542: GLX: No GLXFBConfigs returned
[03:32:51 - 13/04/2025] (E) Glfw Error 65545: GLX: Failed to find a suitable GLXFBConfig
[03:32:51 - 13/04/2025] (W) Could not init GLFW Window; falling back to OpenGL 2.1...
[03:32:51 - 13/04/2025] (E) Glfw Error 65542: GLX: No GLXFBConfigs returned
[03:32:51 - 13/04/2025] (E) Glfw Error 65545: GLX: Failed to find a suitable GLXFBConfig
[03:32:51 - 13/04/2025] (C) Could not init GLFW Window! Exiting
```

nikeedev, 2025-04-13:
Found it