SatDump/SatDump #944: Fengyun2 VISSR Line Corrector

Serenitysalvation, 2025-04-03:
**What would you want to be added to satdump?**
A Fengyun2 VISSR line corrector

**Describe the feature you'd like to have**
A built in feature on SatDump 

**Describe alternatives you've considered**
Tried the corrector available on github but the windows based version fails to initiate and I don't understand python script to use it that way.

**Additional context**

Is it possible to add a Fengyun2 line corrector within SatDump as the program  [fengyun2-svissr-correcto](https://github.com/Foxiks/fengyun2-svissr-corrector) Java based Windows program does not work on my Win11 PC. I do not understand Python script to run it that way...regards Bob

Cpt-Dingus, 2025-05-04:
#934 is pending with this change

Aang23, 2025-05-05:
Merged

closed 2025-05-05 by Aang23

Serenitysalvation, 2025-05-05:
Thankyou to everyone involved in the update enhancement..