SatDump/SatDump #946: Multiple sats transmission decoding at the same time

CharlieDeltaOskar, 2025-04-07:
**What would you want to be added to satdump?**
In case multiple satellites are active on schedule, and SDR bandwidth is wide enough, start decoders for each satellite.

**Describe the feature you'd like to have**
Currently, multiple decoders for one sattelite is supported but no support for multiple satellites. I want to start decoders for each satellite if their schedules are overlapping.

**Additional context**
for NOAA 15-18-19 sattelites sometimes schedules are overlapping and RTL-SDR v4 bandwidth is suffisient to receive all of it, I see signal on waterfall, but I can't add additional decoder.

Aang23, 2025-04-07:
Actually, this can already be done :
https://www.a-centauri.com/articoli/automating-satellite-reception-with-satdump

CharlieDeltaOskar, 2025-04-08:
Do you test this? Because of this schedule, the app will start receiving all channels of NOAA 19, skipping NOAA 18, and begin receiving a part at the end of NOAA 15 after the NOAA 19 session ends.

[image]

CharlieDeltaOskar, 2025-04-08:
Looks like I missed Multi mode. Thank you!

closed 2025-04-08 by CharlieDeltaOskar