SatDump/SatDump #950: MCIR RGB Composite in SatDump 1.2.2 Fails with "requires projections info" Error

ponyjoy, 2025-04-17:
**Description:**

Hello SatDump team,

First, thank you for your amazing work on this project! I'm encountering an issue while using SatDump version 1.2.2 for offline decoding and visualization of NOAA APT data, and I’d greatly appreciate your guidance.

**Issue:**

I’ve successfully decoded APT images from NOAA satellite recordings using SatDump 1.2.2. However, when attempting to create an RGB composite using the "MCIR" option in the Viewer, I consistently receive an error message: "Requires projections info." I’m unsure how to resolve this or what steps are needed to provide the required projection information.

**Steps to Reproduce:**

Load a recorded NOAA APT audio file into SatDump 1.2.2.
Process the audio file to decode the APT image successfully.
Open the decoded image in the Viewer.
Select "RGB Composites" and choose the "MCIR" option.
The error "Requires projections info" appears, and the composite is not generated.

**Environment:**

SatDump Version: 1.2.2
Operating System: Windows 10
Processing Mode: Offline decoding
Input Data: NOAA APT audio recording
Attachments:

To help diagnose the issue, I’ve attached the following files:

A screenshot of the error message displayed in SatDump.

[image]

The original NOAA APT audio recording used for decoding.
[https://drive.google.com/file/d/1emh9q52mwl-lIy8vBnZUBBLtPEsfaol5/view?usp=sharing](url)

**Questions:**

What does the "Requires projections info" error mean, and how can I provide the necessary projection information?
Are there specific settings or additional steps I need to configure in SatDump to enable MCIR composites for NOAA APT data?
Is this a known limitation of the APT decoding pipeline, or am I missing something in my workflow?

**Additional**:

I’m relatively new to satellite image processing, so I apologize if this is a basic question. I’ve checked the documentation and searched the issues tracker but couldn’t find specific guidance on this error. Any advice or pointers to relevant documentation would be greatly appreciated.

Thank you in advance for your time and support! Please let me know if you need any additional information or logs to investigate this issue further.

ponyjoy, 2025-04-17:
I have an update that might help with this issue. I tested the same NOAA APT audio file on another computer running SatDump 1.2.2. On this second computer, the Viewer’s Products tab includes an additional Projection option (distinct from the Projections tab) alongside Image, RGB Composite, and Map Overlay. Using this Projection option, I successfully generated the MCIR RGB composite without the "Requires projections info" error.

On the problematic computer, the Projection option is missing from the Products tab. Both systems use SatDump 1.2.2, but the operating systems are Windows 11 on the problematic computer,. The workflow and input file were identical.

Could the missing Projection option be causing the error? Are there specific settings or dependencies needed for it to appear? Please let me know if you need screenshots or logs from either system.

[image]

[image]

Hue-Jhan, 2025-05-05:
idk maybe when u did the offline processing you ticked the Start Timestamp Auto box, that must not be ticked, idk if this may solve the issue

RAD750, 2025-05-12:
Hi, as seen here: 
[image]

your pass has a timestamp of January 1, 1970. This is the default date (start of Unix epoch) set when the pass has no timestamp information. This is usually caused by a simple mistake, e.g. renaming the file to something SatDump can't get date/time information from (we support default timestamped names for SDR++, HDSDR+ SDRUno, GQRX and a few others). 
If you have the date, and time (with seconds accuracy) when you **started** the recording, you can input those in the `Start Timestamp` box in the `Offline Decode` tab.
[image]

Please note that the timestamp is in UTC, so you will probably need to convert your local timezone to UTC. It is also compulsory for the timestamp to be accurate to the second, because the satellite's position is extrapolated from it using orbital elements (TLE) as APT doesn't carry any timestamp information from the satellite itself.

RAD750, 2025-05-12:
Just FYI, usually selecting Auto does the trick and shouldn't cause issues, which means there is some sort of either user error or bug. If you could tell me how you created the audio file and which name it has, it would be very useful.
Thanks!

Hue-Jhan, 2025-05-12:
> @RAD750  @Hue-Jhan  You are right! Your suggestions were absolutely spot on and completely resolved the issue I was facing. It seems the root cause of the error was **not setting the correct timestamp** as you pointed out. Following your advice, everything is now working correctly.
> Thank you again for your help!
> 73!

No problem dude, can you set my comment as official answer plsss? I need it for the GitHub achievement plssss

Hue-Jhan, 2025-05-13:
@ponyjoy pretty pleaseee

ponyjoy, 2025-05-13:
> [@ponyjoy](https://github.com/ponyjoy) pretty pleaseee

Hey buddy, how can I set your answer as the official answer? I don't know how to do it

Hue-Jhan, 2025-05-13:
> > [@ponyjoy](https://github.com/ponyjoy) pretty pleaseee
> 
> Hey buddy, how can I set your answer as the official answer? I don't know how to do it

Idk, maybe the option is on the 3 dots on the right side of the comment, then you should close the issue with the button on the bottom of this page, there is a small arrow on the side of that button, maybe theres an option to set the answer there

RAD750, 2025-05-14:
Dear @ponyjoy 
> I recorded .wav audio files using the Android version of SDR++. The file names are typically in a format like audio_137910000Hz_11-20-53_17-04-2025.

This should never happen, I use SDR++ daily and never had this happen. It would be interesting to know if this happened only once or on all passes. The reason why it happened is because the frequency is wrong. I assume you wanted to record NOAA-18, whose frequency is 137.9125 and not 137.9100. Adjust the frequency and automatic timestamping should work again (it relies on the correct frequency to identify the satellite in question)

For your friend in China look towards the same issue. Additional points worth investigating:
* Outdated TLEs
* Wrong location set (if he uses SatDump to record)
* Wrong satellite selected (especially if using autotrack)

closed 2025-05-25 by ponyjoy