SatDump/SatDump #952: Network Connection Issue

njrusty, 2025-04-19:
**Description of the issue**
Reinstalled all sofware due to network dropout, wont connect to RPI4. When I plug in the CATV cable to the RPI4, everything works great. However; when I connect the USB adapter from my GOES antenna (from the LNA) to the RPI4, the network connection lights constantly go out after about 10 seconds.

**Hardware (SDR/PC/OS)**
RPI4 and lastest satdump as of today; 4/19/25.

Best guess: LNA shorting out or something. LNA is actually inside my observatory so it is not exposed to the elements.

njrusty, 2025-04-21:
Replaced the LNA. still cannot connect with network. Light go on and start blinking then the lights go out for about 10 seconds then repeats.

NicolasGagne, 2025-08-06:
Similar happen to me my SDR dongle was shorted out and it would cause the Raspberry pi to reboot. 
Have you try the SDR dongle on an other computer/RPI?

closed 2026-02-25 by Aang23