SatDump/SatDump #956: Installation Issue: unable to unzip?

njrusty, 2025-04-23:
**Description of the issue**
On USRADIOGUY installation of Satdump on RPI4, got to step to execute the following:
sudo apt install ./satdump_rpi64_latest_arm64.deb

But keep getting the following error:

E: read, still have 8 to read but none left
E: Internal error, could not locate member control.tar{.zst,.lz4,.gz,.xz,.bz2,.lzma,}
E: Could not read meta data from /home/goes/satdump/satdump_rpi64_latest_arm64.deb
E: The package lists or status file could not be parsed or opened.

If I execute 'ls' in the Satdump directory, all I see is the file:
./satdump_rpi64_latest_arm64.deb

**Hardware (SDR/PC/OS)**
RPI4 running latest Linux

**Version (Eg, 1.0.0, CI Build 171)**

njrusty, 2025-04-23:
I also noticed the following error:

sudo systemctl status satdump-goes
Unit satdump-goes.service could not be found.

RAD750, 2025-05-12:
Where did you obtain the debfile? It seems like it is corrupted or damaged.