SatDump/SatDump #958: Installation Issues!!!!

njrusty, 2025-04-26:
**Description of the issue**
LITERALLY another 6 hours wasted again today trying to get SatDump to work. 
Went through 4 installations following instructions from the readme file on the site here and instruction from usradioguy.com site and each case, it appears the plugins folder is not being loaded.

I've tried to manually load it in the usr folder but cannot break through the permission to allow me load.

Any help will be appreciated. Put satellite dish into concrete to support antenna and now thinking of demolishing everything. 

**Hardware (SDR/PC/OS)**
RPI4

**Version (Eg, 1.0.0, CI Build 171)**

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

Cpt-Dingus, 2025-05-04:
Can you share the exact error you're seeing?

The README should have everything you need, how are you installing and running the app?

njrusty, 2025-05-04:
Will post when I get home. I have followed the readme directions exactly  times and still not getting the plugin folder to load.

njrusty, 2025-05-08:
Trying installation again (7th time now) and getting the attached error.

[image]

Cpt-Dingus, 2025-05-08:
did you install the libglfw3-dev dependency

njrusty, 2025-05-08:
I don't know. I followed the instructions as provided 

Yahoo Mail: Search, Organize, Conquer

Cpt-Dingus, 2025-05-08:
Try adding -DBUILD_GUI=OFF to the cmake command

njrusty, 2025-05-08:
Thank you, will try

Yahoo Mail: Search, Organize, Conquer

njrusty, 2025-05-08:
> Installing the dependencies is the first step, glfw3 should be installed
>
> image.png (view on web)

Yes, I followed that step. But on the 'install libvolk-dev, that gave an error so I did 'install libvolk2-dev'.
Regards,Russ

RAD750, 2025-05-12:
libvolk and libvolk2 are interchangeable, don't worry