SatDump/SatDump #959: Unable to build on Fedora 42

ethanbowering24, 2025-04-28:
I'm not sure what's causing this. I installed SatDump fine on 41 but have been having some issues since upgrading to 42 and wanted to try rebuilding and reinstalling.

```In file included from /home/ethanbowering/Documents/radio/SatDump/src-core/common/image/composite.cpp:7:
/home/ethanbowering/Documents/radio/SatDump/src-core/./libs/sol2/sol.hpp: In member function ‘T& sol::optional<T&>::emplace(Args&& ...)’:
/home/ethanbowering/Documents/radio/SatDump/src-core/./libs/sol2/sol.hpp:6755:31: error: ‘class sol::optional<T&>’ has no member named ‘construct’ [-Wtemplate-body]
 6755 |                         this->construct(std::forward<Args>(args)...);
      |                               ^~~~~~~~~
make[2]: *** [src-core/CMakeFiles/satdump_core.dir/build.make:1465: src-core/CMakeFiles/satdump_core.dir/common/image/composite.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/ethanbowering/Documents/radio/SatDump/src-core/common/lua/lua_utils.h:3,
                 from /home/ethanbowering/Documents/radio/SatDump/src-core/common/lua/lua_utils.cpp:1:
/home/ethanbowering/Documents/radio/SatDump/src-core/./libs/sol2/sol.hpp: In member function ‘T& sol::optional<T&>::emplace(Args&& ...)’:
/home/ethanbowering/Documents/radio/SatDump/src-core/./libs/sol2/sol.hpp:6755:31: error: ‘class sol::optional<T&>’ has no member named ‘construct’ [-Wtemplate-body]
 6755 |                         this->construct(std::forward<Args>(args)...);
      |                               ^~~~~~~~~
make[2]: *** [src-core/CMakeFiles/satdump_core.dir/build.make:1829: src-core/CMakeFiles/satdump_core.dir/common/lua/lua_utils.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1633: src-core/CMakeFiles/satdump_core.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

arobertson404, 2025-04-29:
Have the same issue with Fedora 42 and managed to get it to compile by:

sudo dnf install airspyone_host-devel armadillo-devel cmake curl-devel fftw-devel g++ git glew-devel glfw-devel hackrf-devel hdf5-devel intel-opencl jemalloc-devel libomp-devel libpng-devel libtiff-devel libzstd-devel luajit-devel nng-devel ocl-icd ocl-icd-devel portaudio-devel rtl-sdr-devel volk-devel zenity google-roboto-fonts

*************************

SatDump-master/src-core/libs/sol2/sol.hpp

		template <class... Args>
		T& emplace(Args&&... args) noexcept {
			static_assert(std::is_constructible<T, Args&&...>::value, "T must be constructible with Args");

-			*this = nullopt;
-			this->construct(std::forward<Args>(args)...);
		}

+			this->reset();
+			new (static_cast<void*>(this)) T(std::forward<Args>(args)...);
+			return *this;
		}

*************************

SatDump-master/plugins/inmarsat_support

There were errors with inmarsat, so I just Commented out everything as it doesn't compile.

*************************

satdump-ui to run after install. If you try to run it without installing you get errors with Roboto Medium fonts.

mk0x55, 2025-05-03:
Same here, I managed to solve it simply by commenting out the entire definition of
`template<class... Args> T& emplace(Args&& args) noexcept ...`

That should avoid introducing hidden incorrect behavior - if it broke something, stuff using the definition would simply fail to compile. But it didn't break SatDump.

On the other hand, there were some other errors in Inmarsat as you mentioned - these can be easily fixed by the following changes:
In plugins/inmarsat/aero/libacars/list.h, change the definitions of la_list_foreach, la_list_free_full and la_list_free_full_with_ctx as follows:
`void la_list_foreach(la_list *l, void (*cb)(const void* p, void* c), void *ctx);
void la_list_free_full(la_list *l, void (*node_free)(void* tag));
void la_list_free_full_with_ctx(la_list *l, void (*node_free)(void* tag, void* c), void *ctx);
`
The problem was missing arguments... I wonder how and why it would compile on older systems... I mean even the functions in list.c supplied the arguments that were missing in the header file. To my eyes that is an inconsistency within the libacars-codebase itself - but I might simply be ignorant of a change to the C++ standard or the GCC compiler... someone with enough insight would have to be the judge here.

Finally, make a corresponding adjustment to plugins/inmarsat/aero/libacars/list.c... just adding the arguments, so that it would look like this:
`
void la_list_foreach(la_list *l, void (*cb)(const void* p, void* c), void *ctx) {
        la_assert(cb != NULL);
        for(; l != NULL; l = la_list_next(l)) {
                cb(l->data, ctx);
        }
}

void la_list_free_full_with_ctx(la_list *l, void (*node_free)(void* tag, void* c), void *ctx) {
        if(l == NULL) {
                return;
        }
        la_list_free_full_with_ctx(l->next, node_free, ctx);
        l->next = NULL;
        if(node_free != NULL) {
                node_free(l->data, ctx);
        } else {
                LA_XFREE(l->data);
        }
        LA_XFREE(l);
}

void la_list_free_full(la_list *l, void (*node_free)(void* tag)) {
        if(l == NULL) {
                return;
        }
        la_list_free_full(l->next, node_free);
        l->next = NULL;
        if(node_free != NULL) {
                node_free(l->data);
        } else {
                LA_XFREE(l->data);
        }
        LA_XFREE(l);
}
`

After this, you should be able to compile the project including the Inmarsat plugin.

Vitaspiros, 2025-05-10:
I'm having the same problem on Arch Linux. Maybe it has to do with the compiler version?

MWPSBCID, 2025-05-15:
mk0x55's solution works for me as well. Also on Arch.

nikeedev, 2025-06-09:
@Aang23, sorry for ping, but we need to add [this fix](https://github.com/SatDump/SatDump/issues/959#issuecomment-2840325282), as SatDump would not compile with newer GCC compilers, as the sol2 code is not up-to-date.

I am running GCC 15, and i am also getting similar error:
```cpp
In file included from /home/nikee/apps/SatDump/src-core/common/image/composite.cpp:7:
/home/nikee/apps/SatDump/src-core/./libs/sol2/sol.hpp: In member function ‘T& sol::optional<T&>::emplace(Args&& ...)’:
/home/nikee/apps/SatDump/src-core/./libs/sol2/sol.hpp:6755:31: error: ‘class sol::optional<T&>’ has no member named ‘construct’ [-Wtemplate-body]
 6755 |                         this->construct(std::forward<Args>(args)...);
      |                               ^~~~~~~~~
```