SatDump/SatDump #967: Problem using the Offline Decoder with EUMETSAT file

F5AHD1, 2025-05-09:
[satdump-20250507T231322.log](https://github.com/user-attachments/files/20114958/satdump-20250507T231322.log)

[satdump-20250507T231322.log](https://github.com/user-attachments/files/20114968/satdump-20250507T231322.log)

I downloaded an AHVRR Level 1B Metop Global file following the recommendations on the pages of Jacopo's Lair, one of the developers of satdump.

Site EUMETSAT : https://data.eumetsat.int/data/map/EO:EUM:DAT:METOP:AVHRRL1?start=2025-05-07T00:19:00.000Z&end=2025-05-09T02:19:03.000Z&page=2,20&sort=start,time,0

Downloaded file : AVHR_xxx_1B_M03_20250507085803Z_20250507104003Z_N_O_20250507103510Z
The time slot 08:58 to 10:40

The program decodes for a few hours and enters an indefinite state, see attached protocol.

As this mode of operation is in experimental mode, I wanted to give you this feedback

Hoping you can reproduce the error

**Hardware : Windows 10, 64 bits

**Version : Nightly build as recommended

André F5AHD

RAD750, 2025-05-12:
Hi André, first of all thanks for reading my guide!
Unfortunately I am unable to reproduce this issue on my machine (on the Linux one). I thought it could perhaps be a Windows-only issue, but I was unable to reproduce it on a Windows 10 machine too.

Given the log, I simply think that you are attempting to automatically process such a large pass on an underpowered machine, which could take quite a while. I strongly suggest trying to disable the `Automatically process products` option (`Settings -> General SatDump -> Automatically process products`) and then use the `Viewer` to process only what you need.

Let me know if it solves it!

F5AHD1, 2025-05-12:
[18:46:30 - 12/05/2025] (E) Usage : C:\SatDump\bin\satdump-ui.exe [downlink] [input_level] [input_file] [output_file_or_directory] [additional options as required]
[18:46:30 - 12/05/2025] (E) Extra options (examples. Any parameter used in modules can be used here) :
[18:46:30 - 12/05/2025] (E)  --samplerate [baseband_samplerate] --baseband_format [cf32/cs32/cs16/cs8/wav16] --dc_block --iq_swap
[18:46:30 - 12/05/2025] (D) Starting with OpenGL 4.6.0 NVIDIA 391.35
[18:46:30 - 12/05/2025] (D) Max texture size: 16384
[18:46:30 - 12/05/2025] (I)    _____       __  ____                      
[18:46:30 - 12/05/2025] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[18:46:30 - 12/05/2025] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[18:46:30 - 12/05/2025] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[18:46:30 - 12/05/2025] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[18:46:30 - 12/05/2025] (I)                                     /_/      
[18:46:30 - 12/05/2025] (I) Starting SatDump v1.2.3-c3ec0bd
[18:46:30 - 12/05/2025] (I) 
[18:46:30 - 12/05/2025] (I) Loading config C:\SatDump\share\satdump\satdump_cfg.json
[18:46:30 - 12/05/2025] (I) Loading user settings C:\Users\andre\AppData\Roaming/satdump/settings.json
[18:46:30 - 12/05/2025] (I) Loading plugins from C:\SatDump\lib\satdump\plugins
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\aaronia_sdr_support.dll...
[18:46:30 - 12/05/2025] (E) Error loading C:\SatDump\lib\satdump\plugins\aaronia_sdr_support.dll! Error : "C:\SatDump\lib\satdump\plugins\aaronia_sdr_support.dll": Le module spécifié est introuvable. => src-core\core\plugin.cpp:19
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\aim_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin aim_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\airspyhf_sdr_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin airspyhf_sdr_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\airspy_sdr_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin airspy_sdr_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\analog_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin analog_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\aws_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin aws_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\bladerf_sdr_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin bladerf_sdr_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\bluewalker3_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin bluewalker3_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\cluster_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin cluster_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\cpp_compos_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin standard_cpp_compos loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\cubesat_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin cubesat_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\dmsp_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin dmsp_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\dscovr_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin dscovr_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\dvb_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin dvb_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\earthcare_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin earthcare_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\elektro_arktika_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin elektro_arktika_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\eos_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin eos_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\fengyun2_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin fengyun2_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\fengyun3_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin fengyun3_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\fengyun4_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin fenyun4_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\gcom_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin gcom_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\geonetcast_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin geonetcast_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\gk2a_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin gk2a_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\goes_support.dll...
[18:46:30 - 12/05/2025] (T) Plugin goes_support loaded!
[18:46:30 - 12/05/2025] (T) Loading plugin C:\SatDump\lib\satdump\plugins\hackrf_sdr_support.dll...
[18:46:31 - 12/05/2025] (T) Plugin hackrf_sdr_support loaded!
[... cut ...]
[18:57:09 - 12/05/2025] (D) Workgroup size 4096
[18:59:21 - 12/05/2025] (D) GPU Processing Time 1.325532
[18:59:22 - 12/05/2025] (D) Using GPU! Double precision requested 0
[18:59:22 - 12/05/2025] (T) OpenCL context already initilized.
[18:59:22 - 12/05/2025] (D) Workgroup size 4096
[19:04:04 - 12/05/2025] (D) GPU Processing Time 2.819422
[19:04:06 - 12/05/2025] (D) Using GPU! Double precision requested 0
[19:04:06 - 12/05/2025] (T) OpenCL context already initilized.
[19:04:06 - 12/05/2025] (D) Workgroup size 4096

RAD750, 2025-05-12:
It looks like you didn't disable that option, as it is still trying to project the images.

F5AHD1, 2025-05-12:
HelloI've just checked, the option is disabled, as shown in the screenshot in the appendix.André F5AHDenvoyé : 12 mai 2025 à 22:00de : Jacopo ***@***.***>à : SatDump/SatDump ***@***.***>cc : F5AHD1 ***@***.***>, Author ***@***.***>objet : Re: [SatDump/SatDump] Problem using the Offline Decoder with EUMETSAT file (Issue #967)RAD750 left a comment (SatDump/SatDump#967)It looks like you didn't disable that option, as it is still trying to project the images.

RAD750, 2025-05-14:
Are you trying to project manually then? It is obviously doing something with the projection, as shown here
> [18:50:14 - 12/05/2025] (I) Update projection...
[18:50:14 - 12/05/2025] (I) Reprojecting layer 0...

F5AHD1, 2025-05-14:
HelloYes I tried to make a projection, after getting the image very quickly.I'm very interested in HRPT reception.But as i don't have all the equipment yet, i am struggling with EUMETSAT downloads to familiarize myself with satdump.I haven't yet understood all the possibilities of this fabulous program.And it's not impossible that i have misunderstood something.You are a professional metaeorologist who is generous enough to make his knowledge available to the general public.I am just a radioamateur using raspinoaa for NOAA's APT modes.And now i want to investigate HRPT a little further.André F5AHDenvoyé : 14 mai 2025 à 09:17de : Jacopo ***@***.***>à : SatDump/SatDump ***@***.***>cc : F5AHD1 ***@***.***>, Author ***@***.***>objet : Re: [SatDump/SatDump] Problem using the Offline Decoder with EUMETSAT file (Issue #967)RAD750 left a comment (SatDump/SatDump#967)Are you trying to project manually then? It is obviously doing something with the projection, as shown here[18:50:14 - 12/05/2025] (I) Update projection... [18:50:14 - 12/05/2025] (I) Reprojecting layer 0...

RAD750, 2025-05-17:
Hi, I understand, let me explain.
Projections are a computationally intensive task, especially if processing with the GPU is not enabled. In your case, you are trying to project a very big image on quite a weak computer, which can take a very long time. When projecting, make sure that your projection image size (circled below) is small, not more than 2048x2048, as both memory and computation time quadruples for each doubling in the image size.
With the settings below (equirectangular, both Auto boxes checked, and 0.3 m/px as resolution) it will result in a 1200x600 image which should be generated with your CPU in less than 10 minutes.
[image]