SatDump/SatDump #985: Could not find a handler for source type : rtlsdr! (only in NOAA) - Satdump Live

ja-Bertolin, 2025-06-11:
**Description of the issue**
I am using RN2 NOAA. SATDUMP with this syntax

case "$RECEIVER_TYPE" in
     "rtlsdr")
         samplerate="1024000"
         receiver="rtlsdr"
         decimation=25
         ;;
$SATDUMP live noaa_apt "$audio_temporary_storage_directory" \
    --source "$receiver" \
    --samplerate $samplerate \
    --frequency "${NOAA_FREQUENCY}e6" \
    --gain $gain_option $GAIN $bias_tee_option $crop_topbottom \
    --start_timestamp $PASS_START $finish_processing \
    --satellite_number ${SAT_NUMBER} --timeout $CAPTURE_TIME  >> $NOAA_LOG 2>&1

produces the error "Could not find a handler for source type : rtlsdr!" when $receiver = rtlsdr however the same structure for METEOR works fine

$SATDUMP live meteor_m2-x_lrpt${mode} "$audio_temporary_storage_directory" \
    --source $receiver  \
    --samplerate $samplerate \
    --frequency "${METEOR_FREQUENCY}e6" \
    --gain $gain_option $GAIN $bias_tee_option $finish_processing \
    --baseband_output "$audio_temporary_storage_directory/meteor_m2-x_lrpt${mode}.cadu" \
    --timeout "$CAPTURE_TIME"  >> $NOAA_LOG 2>&1

**Hardware (SDR/PC/OS)**

RPi 4 Bullseye

**Version (Eg, 1.0.0, CI Build 171)**
1.2.3

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

[satdump-20250611T230208.log](https://github.com/user-attachments/files/20698163/satdump-20250611T230208.log)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

Aang23, 2025-06-22:
I cannot reproduce that on my end, you're probably better off forwarding the issue to RN2 unfortunately...

Aang23, 2025-06-22:
I'll close this for now - but tag me if you need anything else related to this issue.