SatDump/SatDump #1011: GK-2A HRIT: Interface lock and LRIT CRC is invalid when .png data is partially written to disk

sergsero, 2025-08-02:
GK-2A, L-band HRIT 1695.4 MHz, i9-13900K/64G/4060Ti-16G/Ubuntu 24.04

When SatDump automatically recording part of the received data as .png files to disk, the interface and receiver functions become blocked. This mostly happens when recording from the VI006 channel.

This causes an error in the decoder and get message : "LRIT CRC is invalid, but the file can be recovered",
which results in part of the image being damaged in the form of a stripe.

[image]

[image]

The problem exists in both versions, 1.2.3 and 2.x.x.

Cpt-Dingus, 2025-08-11:
This is your computer not being able to handle both saving imagery and demodulating new one coming in concurrently - the little spikes are dropped samples. Record baseband, modify power options or process affinity, or use a more powerful machine. Gk2a imagery is rather large!

sergsero, 2025-08-11:
Thank you for your response.
Although it may be the only option, pre-recording a .cadu file and its subsequent processing would be extremely inconvenient.
Unfortunately, my computer's power is limited to an i9-14900K CPU. The drives are PCIe 4.0 Samsung 980/990pro M.2 SSDs only.
The process of affinity seems like an interesting idea.

Derisis13, 2025-10-03:
Hi, I'm facing the same problem, but with L3 LRIT. I've noticed that the FFT and decoder were fine while running on their own, but when I started recoding, it started to have noticable hickups. When I turned off recording, it stopped having them.
By hickups I mean a split seconds of freeze on the FFT, waterfall, and constellation graphs.
The hardware we ran satdump on is relatively low-end (decade old laptop with 4ht gen i5 + 8GB RAM, sata SSD), but I also experienced similar behavior on my main laptop (8th gen i5, 16GB RAM, nvme SSD) but only when writing images.

I wonder if using a temp folder would improve performance (as it'd avoid disk IO)

Derisis13, 2025-10-03:
[image]
This is how the dropped samples affected our image... Very similar, just better recovery

Aybarnur, 2025-10-03:
Despite all the suggestions, for which I am very thankful to everyone who has responded, my computer locking issue GK-2A HRIT persists. Probably really the computer not being able to handle both saving imagery and demodulating new one coming in concurrently.
At the same time, the computer's CPU i9-13900K/RAM 64Gb/GPU 4060Ti-16Gb and Ubuntu 24.04 OS. The type of recording drive that would be used: SSD Samsung 980 PRO NVMe M.2 1Tb (PCIe Gen 4.0 x4).
Now I don't have a more powerful platform, so I was unable to check and the problem of freezing and loss still exists.

Cpt-Dingus, 2025-10-03:
@Derisis13 Hi, you should try just recording baseband and decoding after the fact - it is much less taxing on resources. Otherwise the best you can do is use a lower sampling rate, switch to the high performance power plan, set satdump to the highest priority, or close all background programs. 

Can you send a screenshot of when this happens?

sergsero, 2025-10-04:
@Cpt-Dingus Hello, thank you for your response! I will be waiting for a new DSP, in the hope that it will solve my problem.

Derisis13, 2025-10-05:
> Can you send a screenshot of when this happens?

You mean a video, right? The lockups themeslves are for a few hundred milliseconds (my guess is like .5s, but I'm no atomic clock), just enough to drop a few samples. 

Sadly L3 LRIT is off for days now, but I have recordings that are missing the dropped samples