SatDump/SatDump #1018: Meteosat Third Generation (MTG)

schmiernippel, 2025-08-08:
Receiving MTG Images would be really nice! 
Thanks for the great software!

greets

Aang23, 2025-08-08:
Hello!

This is currently only possible by either downloading the data from the internet (via the EUMETSAT data store or the loader in SatDump), or via EUMETCast. In all cases I recommend using the `verywip`/v2.0.0 of SatDump which has much better support than the current release. In that version, you can just drag any MTG file from EUMETCast directly on SatDump and it'll work (and if not, please report - it's a bug).

Unfortunately, receiving raw data directly like SatDump can on MSG is not possible, as they decided to encrypt the downlink (even though it's... Public data). It's proving to be a real problem as their own stations are extremely unreliable - breaking about once a week.

You can see an attempt a few others & myself had at decoding the data on 26Ghz : https://x.com/aang254/status/1940082276903145812

Aang23, 2025-08-08:
That's the idea... You're only supposed to pay for it, not use it. Too bad if it could be useful in emergency situations, education... etc. :-)

At least the russians still give us ELEKTRO... Which to be honest for now has an imager is better shape than FCI.