SatDump/SatDump #1034: Implement Lime XTRX for way higher bandwidths (120 MHz)

DL7AD, 2025-09-04:
Hi,

I'd like to implement the [Lime XTRX SDR](https://limemicro.com/sdr/limesdr-xtrx/) to support higher bandwidths. Technically it's a LimeSDR but it's not supported by the original Lime Suite. It's only supported by the [Lime Suite NG](https://github.com/myriadrf/LimeSuiteNG). I integrated Lime Suite NG for Satdump on my Fork and it seems to work well up to 120 MHz bandwidth. On the input I operate a S-/X-/Ka-Band dish.

[image]

I want to merge my fork at some time, but only once I've verified, the implementation works without issues. I integrated the plugin in a way that it support both libs at the same time - Lime Suite and Lime Suite NG. If both libs are installed, Lime Suite seems to claim all devices except the ones that it doesn't support (only XTRX atm). The XTRX SDR is then claimed by Lime Suite NG. So there are no duplicate entries in the SDR dropdown list.

https://github.com/DL7AD/SatDump/tree/master/plugins/sdr_sources/lime_xtrx_sdr_support

The (O)QPSK constellation looks good, however the deframer does't like the signal. So far, I've tested it with AURA and another sat signal, that is ~56 MBaud QPSK (Standard CCSDS 131.0).

Could someone verify that the data format looks right:

https://github.com/DL7AD/SatDump/blob/master/plugins/sdr_sources/lime_xtrx_sdr_support/lime_xtrx_sdr.cpp#L128

... and that I didn't mess up with the data conversion:

https://github.com/DL7AD/SatDump/blob/master/plugins/sdr_sources/lime_xtrx_sdr_support/lime_xtrx_sdr.cpp#L273

... or are the IQ frames chopped in the stream? But it also doesn't decode when I reduce the SDR to lower sample rates like 30 Msps. I'm currently out of ideas.

Does anybody have advise?

__AURA satellite 8160 MHz, 7.5 MBaud, OQPSK, Deframer unstable (should be stable at SNR 12dB)__
[image]

__56 MBaud, QPSK, deframer unstable, gibberish data on the CADU output__
[image]

Sven

DL7AD, 2025-09-05:
Additional Info. I seems like the data is chopped. This is the pass by AQUA, where good data can be seen but every second packet is lost. I tried to increase the buffer size from `sample_rate/250` to `sample_rate/25` but it didn't help in any way.

__Europe (upsidedown)__
[image]

__South UK__
[image]

DL7AD, 2025-09-24:
I tried that a week ago (just haven't it posted here yet).

It works this way. Say recording and then post processing the data. I still wonder why that is because if I run the same with a BladeRF, I don't run into this trouble. However I have to say that this computer doesn't have a dedicated graphics card. It runs on a i7-14700 that has the internal UHD770 Graphics only. No idea if that is a problem.

Sven

Aang23, 2026-09-30:
This should be working now.

closed 2026-09-30 by Aang23